tiny-pilot/tinypilot #1253: Disk size requirement

pacmac, 2023-01-07:
I am installing this on a bananapi zero with 16GB SSD.

I realise that this is not a supported platform and are willing to work through the issues to get it working given the unavailability of raspberry Pi's now.

But I am getting the following error:

```
ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device
```

I have 13GB of free space, is it possible to clarify what the minimum size is and also maybe check this in the install script before it starts

:-)

Many Thanks

pacmac, 2023-01-08:
I just switched to a 128GB Drive, with 127GB Free and still get an error:

```
...
...
  Downloading pip-22.3.1-py3-none-any.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 2.8 MB/s
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-22.3.1
+ pip install -r requirements.txt
Collecting ansible==2.10.7
  Downloading ansible-2.10.7.tar.gz (29.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 29.9/29.9 MB 1.9 MB/s eta 0:00:00
ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device

WARNING: There was an error checking the latest version of pip.
+ clean_up
+ rm -rf /var/tmp/tmp.ZEJQTD1pEz.yml
+ clean_up
+ rm -rf /var/tmp/tmp.D7EGRtTZ4c.tgz
```

pacmac, 2023-01-09:
abandoned due to hardware incompatability

closed 2023-01-09 by pacmac

pacmac, 2023-01-09:
Thanks;

I was able to overcome the issues and the build was successful.

The problem was that on the banana pi, it was attempting to build everything from source as there were no pre-built binary wheels.

Fortunately the piwheels binaries are compatible and simply adding this solved the problem:

```
echo -e "[global]\nextra-index-url=https://www.piwheels.org/simple\n" > /etc/pip.conf
```
on RPI, this pip.conf file is created automagically when you install Python3, but on bananapi it is not.

The hardware issue was that the bananapi (and orangepi) use paralell CSI port (24 pin) and the PI uses serial CSI (22 pin) and they are not compatible, and therefore the CSI > HDMI board cannot be used.

So that was a show stopper, which was a real shame as I spent a LONG time figuring out the build issues :-)

mtlynch, 2023-01-09:
Ah, thanks for digging into it! Sorry that it turned out to not be compatible with bananapi.

If you haven't seen it, there's a discussion thread of TinyPilot on alternate hardware: https://github.com/tiny-pilot/tinypilot/discussions/755