tiny-pilot/tinypilot #1266: Update fails with ansible hash mismatch

pacmac, 2023-01-18:
Just tried updating from the clean install that I did a few days ago and it fails as below.

```
/opt/tinypilot-privileged/scripts/update
...
...
...
++ hash -r
+ pip install 'pip>=21.3.1'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip>=21.3.1
  Using cached pip-22.3.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-22.3.1
+ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible==2.10.7
  Using cached https://www.piwheels.org/simple/ansible/ansible-2.10.7-py3-none-any.whl (48.2 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    ansible==2.10.7 from https://www.piwheels.org/simple/ansible/ansible-2.10.7-py3-none-any.whl#sha256=4d43bd3c1f639ef1ce1b6b0696885e7e5b4b83aeffda642ffb0bdf3480b202bf (from -r requirements.txt (line 5)):
        Expected sha256 4d43bd3c1f639ef1ce1b6b0696885e7e5b4b83aeffda642ffb0bdf3480b202bf
             Got        240d00fc80aaf3aaaa87c3b30e8777cca84b86bc8ddd80696a0b4807f769f407

+ clean_up
+ rm -rf /var/tmp/tmp.bTT82SerFA.yml
+ clean_up
+ rm -rf /var/tmp/tmp.leF9K3y30l.tgz
```

Not sure why that should be, maybe you can add a --no-cache-dir option in the install / update script ?

pacmac, 2023-01-18:
Ok, sorry but I have solved it:

```
rm -rf ~/.cache/pip
/opt/tinypilot-privileged/scripts/update
```

pacmac, 2023-01-18:
Oh, I spoke too soon:

```
Installing collected packages: PyYAML, pyparsing, pycparser, MarkupSafe, packaging, Jinja2, cffi, cryptography, ansible-base, ansible
Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 PyYAML-6.0 ansible-2.10.7 ansible-base-2.10.17 cffi-1.15.1 cryptography-37.0.4 packaging-21.3 pycparser-2.21 pyparsing-3.0.9
+ ansible-playbook --inventory localhost, install.yml --extra-vars @/var/tmp/tmp.mDlrYpVldb.yml --extra-vars tinypilot_debian_package_path=/opt/tinypilot-updater/tinypilot_20230117153150_all.deb

PLAY [localhost] **************************************************************************************************************************************************************************

TASK [Gathering Facts] ********************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"ansible_facts": {}, "changed": false, "failed_modules": {"ansible.legacy.setup": {"failed": true, "module_stderr": "  File \"/root/.ansible/tmp/ansible-tmp-1674019716.3941934-1023-47005067204228/AnsiballZ_setup.py\", line 26\n    AAAAAAAAAAAAgAFZqwIAYW5zaWJsZS9tb2R1bGVfdXRpbHMvcGFyc2luZy9jb252ZXJ0X2Jvb2wucHlQSwECFAMUAAAACACTKzJWAAAAAAIAAAAAAAAAKAAAAAAAAAAAAAAAgAG6rQIAYW5zaWJsZS9tb2R1bGVfdXRpbHMvcGFyc2luZy9fX2luaXRfXy5weVBLAQIUAxQAAAAIAJMrMlYeETrnOwcAAF4QAAAiAAAAAAAAAAAAAACAAQKuAgBhbnNpYmxlL21vZHVsZV91dGlscy9weWNvbXBhdDI0LnB5UEsBAhQDFAAAAAgAkysyVmteCvHFHwAAdIAAACQAAAAAAAAAAAAAAIABfbUCAGFuc2libGUvbW9kdWxlX3V0aWxzL3NpeC9fX2luaXRfXy5weVBLAQIUAxQAAAAIAJMrMlaKxYWp7wsAAO8hAAAYAAAAAAAAAAAAAACAAYTVAgBhbnNpYmxlL21vZHVsZXMvc2V0dXAucHlQSwECFAMUAAAACACTKzJWAAAAAAIAAAAAAAAAGwAAAAAAAAAAAAAAgAGp4QIAYW5zaWJsZS9tb2R1bGVzL19faW5pdF9fLnB5UEsFBgAAAABiAGIAjiEAAOThAgAAAA==\"\"\"\n                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               ^\nSyntaxError: (unicode error) 'utf-8' codec can't decode byte 0x80 in position 221157: invalid start byte\n", "module_stdout": "", "msg": "MODULE FAILURE\nSee stdout/stderr for the exact error", "rc": 1}}, "msg": "The following modules failed to execute: ansible.legacy.setup\n"}

PLAY RECAP ********************************************************************************************************************************************************************************
localhost                  : ok=0    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0
```

pacmac, 2023-01-18:
Closing this, sorry to waste your time.

Corrupt file system, fresh install

closed 2023-01-18 by pacmac

pacmac, 2023-01-18:
As a side note, I just added a zero case to printables: https://www.printables.com/model/372892-tiny-pilot-zero-case

mtlynch, 2023-01-18:
Thanks for the update! Yeah, that looked like corrupt filesystem. You can enable read-only filesystem if you'd like to make TinyPilot a little more resistant to unexpected power cuts:

https://tinypilotkvm.com/faq/read-only-filesystem

>As a side note, I just added a zero case to printables: https://www.printables.com/model/372892-tiny-pilot-zero-case

Wow, that's awesome! Thanks for sharing that.

How is performance on the Zero? Is it original Zero or Zero 2/2W?

pacmac, 2023-01-18:
Performance is very useable, I dont really notice any lag, I've not run it on one of my PI4s's and I see no need for the extra size, and PI4s are very scarce now, well in fact all PIs are.

I almost had it working on a banana PI, but the limitation there is that it has a Parallell CSI Port and the HDMI > CSI Camera board work work, and can't uses a USB Capture card as the one OTG port cannot work in master / slave mode at the same time.

See attached graph, memory usage is OK, 3 of the cores are always busy with gstreamer (not sure why it needs so many threads) but the other core is idle.

I thought it would get too hot and warp the PLA case, it's warm but not hot.

[image]

mtlynch, 2023-01-19:
Oh wow, that's cool! Impressive that it performs so well.