tiny-pilot/tinypilot #1271: Multiple WebRTC media tracks: when to fallback to MJPEG?

jdeanwallace, 2023-01-20:
As @jotaen4tinypilot pointed, now that we accept multiple WebRTC media tracks, when is the best time to fallback to the MJPEG stream? 

For example, do we stay in H.264 mode, even if we lost the video signal, but still have audio (for whatever reason)?

mtlynch, 2023-01-20:
We should fail over to MJPEG when we lose video. If we lose audio, we stay in H.264 mode because it's still better than MJPEG mode.

What's the current behavior?

jdeanwallace, 2023-01-20:
@mtlynch - Current behaviour is that [if we lose any track (audio or video), we fallback to MJPEG](https://github.com/tiny-pilot/tinypilot/blob/0013065632ba9265083ef452781eb5e81270f138/app/static/js/webrtc-video.js#L177-L180).

My recent [PR to address a smoother WebRTC start-up](https://github.com/tiny-pilot/tinypilot/pull/1270) (still unmerged), unintentionally changes the behaviour to only fallback to MJPEG if we lose all tracks (audio and video)

closed 2023-01-24 by jdeanwallace