tiny-pilot/tinypilot #1279: Retain update logs for longer periods of time

db39, 2023-01-25:
Users sometimes reach out with issues after updating their software and often the update logs are absent because TinyPilot doesn't retain them long enough - making it harder to diagnose issues.

Can we retain the TinyPilot update logs longer?

cghague, 2023-01-26:
I've started looking into this and it seems like the update process is managed by `systemd` and so the logging is handled by `journald`, with the `collect-debug-logs` script retrieving the logs by calling `journalctl`.

I've done some digging and it seems like `journald` on Raspberry Pi OS (and, by the look of it, Debian and Ubuntu too) defaults to discarding logs automatically on each reboot. This seems like a bit of a design flaw to me given that unexpected reboots are often a reason to go looking for logs but presumably there is a reason for it.

We can [configure](https://www.freedesktop.org/software/systemd/man/journald.conf.html) `journald` to use persistent logging by setting `Storage= persistent` in `/etc/systemd/journald.conf`. However, this will affect all `journald` logs and I suspect it will drastically increase the amount of microSD writes that take place.

@mtlynch - I'd suggest a better option might be to add a step to the end of the upgrade process that saves the update logs to a file and have the `collect-debug-logs` script grab that instead. Let me know your thoughts!

mtlynch, 2023-01-26:
Is it possible to adjust storage settings on a per-service basis [like this](https://askubuntu.com/a/659268/322075)? Or is storage not one of the settings we can adjust?

cghague, 2023-01-26:
Unfortunately, the per-service `systemd` overrides don't cover `journald`.

I spent a while trying to find out if there was a way around this earlier, but, despite finding [other people asking the same sort of question](https://unix.stackexchange.com/questions/512569/customize-journald-config-for-a-single-service), I came up empty. It seems that per-service configuration of `journald` just isn't a supported feature.

mtlynch, 2023-01-27:
Yeah, then I agree with your suggestion to persist the update logs to a separate file and then retrieve that from the debug logs.

cghague, 2023-01-30:
I've been looking into this in more depth because I was sure it must be possible. I verified that `journald` is configured to only log to memory by default, which means that any logs that aren't captured using another mechanism are lost upon reboot.

It isn't possible to change this behavior for an individual service, but some research suggested using log namespaces as a workaround. I tried to implement this unsuccessfully. I eventually realized that the version of `journald` that ships with Buster is outdated and doesn't support log namespaces.

I was able to confirm that the version of `journald` included in Bullseye does support log namespaces. I downloaded one of the Bullseye builds of TinyPilot Pro from CircleCI, intending to use log namespaces to enable persistent logging of just the `tinypilot-updater` service. However, this appears unnecessary as `journald` under Bullseye is configured to use persistent logs by default.

This means that the update logs will be retained for longer once we move to Bullseye. The default is to keep up to 100 log files for up to 1 month. We could potentially use log namespaces to extend this for the `tinypilot-updater` service, but I doubt it's needed.

---

@mtlynch - TL;DR: I think this will fix itself when we update the base image to Bullseye. Let me know how you'd like to proceed with this issue (if at all). Thanks!

mtlynch, 2023-01-31:
>However, this appears unnecessary as journald under Bullseye is configured to use persistent logs by default.

>This means that the update logs will be retained for longer once we move to Bullseye. The default is to keep up to 100 log files for up to 1 month. We could potentially use log namespaces to extend this for the tinypilot-updater service, but I doubt it's needed.

Oh, okay. That's easy enough.

On our Pro images, we should configure it to log to RAM, otherwise I think we increase chances of filesystem corruption with excessive writes. I've filed a bug for that in our Pro repo: https://github.com/tiny-pilot/tinypilot-pro/issues/739

closed 2023-01-31 by mtlynch