tiny-pilot/tinypilot #1305: Add mouse clicks to on-screen keyboard

db39, 2023-02-09:
We've seen a number of users in recent weeks asking about the TinyPilot web interface on mobile devices. Most questions  relate to clicking. We have an [open issue for supporting touch screens](https://github.com/tiny-pilot/tinypilot/issues/270), but can we add an 'on-screen mouse' - perhaps just for left and right mouse clicks?

mtlynch, 2023-02-09:
Can we flesh this out a bit? I'm having trouble understanding the request.

We have an on-screen keyboard to allow users to click keys with their mouse that they can't click with their physical keyboard. If there's an on-screen mouse, how does the user interact with it? What additional functionality does it enable?

AltF5, 2023-03-05:
I would recommend an implementation like Jump Desktop’s mouse

yaniv15, 2023-05-09:
I have integrated mouse clicks and keyboard, which is working perfectly fine on android browser(any) for tinypilot controlling linux os. I noticed that this isn't working well for windows os and on IOS browser. I think I can fix it for windows but sadly I dont own an apple device to debug and fix it. 

I'll push the changes in next few days.

mtlynch, 2023-12-06:
Closing this in favor of https://github.com/tiny-pilot/tinypilot/issues/270