tiny-pilot/tinypilot #1313: NO SIGNAL

s13orlando, 2023-02-16:
## Description
NO DIGNAL ERROR
Briefly summarize the problem you're experiencing.
NO SIGNAL ON SCREEN
## What's the behavior that you expect?

_Your answer_no signal

## What's happening instead? ne
er shows other machine

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on '....'ipNO SIGNAL
3. See error

## Screenshots

_If applicable, add screenshots to help explain your problem._

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`[
](https://logs.tinypilotkvm.com/q4RsvfYB)

mtlynch, 2023-02-16:
Thanks for reporting this!

Can you try the troubleshooting steps here?

https://tinypilotkvm.com/faq/no-signal

mtlynch, 2023-03-15:
Closing due to lack of response

closed 2023-03-15 by mtlynch