tiny-pilot/tinypilot #1319: Underscores cannot be entered.

mitubato, 2023-02-22:
Hello.

I'm using Version: 1.8.1-22 [7329336] and I can't type the underscore.
The keyboard I'm using is 109A.
Version: 1.8.1-22 [7329336]

If you try to enter it, the following will be displayed and it will not actually be entered.
[image]

mtlynch, 2023-02-28:
Thanks for reporting this!

@db39 - Can you assist here?

db39, 2023-02-28:
Hi @mitubato, sorry that you're running into this keyboard input issue.

Just to confirm, is the keyboard layout you're using a Japanese layout - like this one?

[image]

Are you able to enter `_` using TinyPilot's on-screen keyboard? To do this, go to View > Show Keyboard (if it isn't already open), then click `shift` + `_` using your mouse. If that works, you can use the on-screen keyboard as a workaround while we try and figure out the root cause of the issue.

Finally, can you go to [keycode.info](https://www.toptal.com/developers/keycode) and type an underscore? Could you share a screenshot of the output like my example?:

[image]

mitubato, 2023-03-01:
こんにちは@db39

使用しているキー配列は画像のものと同じです。
TinyPilotのスクリーンキーボードでは’_’を入力すると’=’と表示されます。

keycode.infoでは226と出ました。
[image]

db39, 2023-03-02:
Thanks for sharing that information, @mitubato!

Unfortunately, TinyPilot doesn't have support for the `IntlRo` key. And I'm not sure if we'd be able to add support for that key in future due to how TinyPilot currently handles keyboard input.

One option is to paste the character (using Actions > Paste in the TinyPilot web interface menu). But I understand that this is an inconvenient workaround.

I'm sorry that I haven't been able to find a better solution for this. Please let me know if you have any other questions.

mtlynch, 2023-03-02:
I think we should be able to fix this.

@db39 - Can you try configuring an on-screen keyboard on a test machine with the Japanese layout that @mitubato is using and then [use evtest](https://github.com/tiny-pilot/tinypilot/issues/361#issuecomment-1419699354) to print the raw HID codes, then press Shift + IntlRo on the on-screen keyboard?

[image]

If we can capture what HID code that key generates on a Japanese system, then we can add a mapping for IntlRo in [js_to_hid.py](https://github.com/tiny-pilot/tinypilot/blob/master/app/js_to_hid.py).

You can test by sending raw HID codes to the TinyPilot emulated keyboard by echoing to `/dev/hidg0`. For example, this simulates a keypress of the `i` key (HID code = `0x0c`):

```bash
echo -ne "\0\0\xc\0\0\0\0\0" > /dev/hidg0
```

mitubato, 2023-03-03:
@db39 Thank you for checking.

@mtlynch
hello.
I tried to check, but I couldn't find/dev/input/event*.
Do I need to make any settings?

`root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
root@raspberrypi:~#
root@raspberrypi:~# cat /etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2
root@raspberrypi:~#`

`root@raspberrypi:~# vcgencmd get_config int
arm_freq=1500
audio_pwm_mode=514
camera_auto_detect=-1
config_hdmi_boost=5
core_freq=500
core_freq_min=200
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=3
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=250
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage_avs=-17500
pause_burst_frames=1
program_serial_random=1
total_mem=4096
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300`

mtlynch, 2023-03-03:
@mitubato - evtest would run on the test machine, not on the TinyPilot device. We're trying to determine what HID code the target machine sees when the user presses the `IntlRo` button on a Japanese system.

mitubato, 2023-03-03:
@mtlynch 
excuse me.

I got the following result:

echo -ne "\0\0\x89\0\0\0\0\0" > /dev/hidg0
  -> \

Since I didn't know the description to send at the same time as 'Shift', it is '/', but if it can be sent at the same time, it seems that '_' can be entered.

`Testing ... (interrupt to exit)
Event: time 1677841534.837291, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2a
Event: time 1677841534.837291, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1677841534.837291, -------------- SYN_REPORT ------------
Event: time 1677841535.092803, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2a
Event: time 1677841535.092803, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 2
Event: time 1677841535.092803, -------------- SYN_REPORT ------------
Event: time 1677841535.113115, type 4 (EV_MSC), code 4 (MSC_SCAN), value 73
Event: time 1677841535.113115, type 1 (EV_KEY), code 89 (KEY_RO), value 1
Event: time 1677841535.113115, -------------- SYN_REPORT ------------
Event: time 1677841535.283676, type 4 (EV_MSC), code 4 (MSC_SCAN), value 73
Event: time 1677841535.283676, type 1 (EV_KEY), code 89 (KEY_RO), value 0
Event: time 1677841535.283676, -------------- SYN_REPORT ------------
Event: time 1677841535.495541, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2a
Event: time 1677841535.495541, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1677841535.495541, -------------- SYN_REPORT ------------

Event: time 1677841544.337540, type 4 (EV_MSC), code 4 (MSC_SCAN), value 17
Event: time 1677841544.337540, type 1 (EV_KEY), code 23 (KEY_I), value 1
Event: time 1677841544.337540, -------------- SYN_REPORT ------------
Event: time 1677841544.446323, type 4 (EV_MSC), code 4 (MSC_SCAN), value 17
Event: time 1677841544.446323, type 1 (EV_KEY), code 23 (KEY_I), value 0
Event: time 1677841544.446323, -------------- SYN_REPORT ------------
^Croot@Ubuntu:~#`

db39, 2023-03-06:
Thanks for sharing that result @mitubato. I tried to use `echo -ne "\0\0\x89\0\0\0\0\0" > /dev/hidg0` to test this on my device, but it doesn't seem work as expected.

Can you share  `evtest`'s output when you type `_` on your Japanese keyboard again?

mitubato, 2023-03-08:
Yes! Share again.
I think the reason why it didn't go well is that the keyboard layout set on the PC connected to TinyPilot is not for Japanese.

`root@ubuntu:~# ./evtest/evtest /dev/input/event3
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab41
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 56 (KEY_LEFTALT)
    Event code 57 (KEY_SPACE)
    Event code 58 (KEY_CAPSLOCK)
    Event code 59 (KEY_F1)
    Event code 60 (KEY_F2)
[... 89 lines cut ...]
    Event code 2 (LED_SCROLLL) state 0
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1678238373.501312, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2a
Event: time 1678238373.501312, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1678238373.501312, -------------- SYN_REPORT ------------
Event: time 1678238373.634227, type 4 (EV_MSC), code 4 (MSC_SCAN), value 73
Event: time 1678238373.634227, type 1 (EV_KEY), code 89 (KEY_RO), value 1
Event: time 1678238373.634227, -------------- SYN_REPORT ------------
Event: time 1678238373.756448, type 4 (EV_MSC), code 4 (MSC_SCAN), value 73
Event: time 1678238373.756448, type 1 (EV_KEY), code 89 (KEY_RO), value 0
Event: time 1678238373.756448, -------------- SYN_REPORT ------------
Event: time 1678238373.834622, type 4 (EV_MSC), code 4 (MSC_SCAN), value 2a
Event: time 1678238373.834622, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1678238373.834622, -------------- SYN_REPORT ------------
^C
root@ubuntu:~#`
[image]
[image]

db39, 2023-03-09:
Thanks for that information, @mitubato.

There are a couple more tests I'd like to try. 

For these tests, can you make sure your target computer is using a Japanese language layout?

Could you SSH into your TinyPilot and then run the following command:

```bash
echo -ne "\x02\0\x87\0\0\0\0\0" > /dev/hidg0
```

I'm expecting that command to output `＿`. Can you share a screenshot of the output (in a text editor or on keycode.info)?

Then try this command:

```bash
echo -ne "\0\0\x89\0\0\0\0\0" > /dev/hidg0
```

This time, I'm expecting it to output `￥`. Can you share a screenshot of the output of this too?

If they both output the expected characters, I can add them to TinyPilot's supported keycodes.

mitubato, 2023-03-13:
Hi! @db39 
I am late.
We will report the results!
You have successfully entered an underscore!

’echo -ne "\x02\0\x87\0\0\0\0\0" > /dev/hidg0’
[image]

’echo -ne "\0\0\x89\0\0\0\0\0" > /dev/hidg0’
[image]

db39, 2023-03-13:
Fantastic! Thanks so much for helping with that, @mitubato.

I'll create a branch to add those codes to TinyPilot. Once they're merged into the main branch, you should be able to use those keys on your keyboard. I will let you know once we've done that.

closed 2023-03-13 by db39

mitubato, 2023-03-14:
@db39 
thank you very much!
We have confirmed that you can enter underscores in TinyPilot!

db39, 2023-03-14:
No problem, @mitubato, I'm glad we were able to find a solution.

I merged the changes into the repository. So once we release the new version of TinyPilot you should have support for `＿` and `￥` keys on  your hardware keyboard.

If you want to incorporate the changes now, you can edit your files to match [my commit](https://github.com/tiny-pilot/tinypilot/pull/1323/commits/af217affd1f63678deec2d8ee54b55d64089088d).