tiny-pilot/tinypilot #1321: Is H.264 supported with HDMI to USB dongle?

jlian, 2023-03-02:
I checked the blogs, wiki, existing issues, and the forum and see no mention. Which makes it seem like H.264 would work with the dongle?

Asking because:
- PiKVM's H.264 support [requires](https://docs.pikvm.org/webrtc/) the HDMI-CSI bridge
- I've [modified](https://github.com/tiny-pilot/tinypilot/issues/829#issuecomment-1205510550) my TinyPilot installation a bit and so hesitant to disturb a working custom setup

mtlynch, 2023-03-03:
It does work, but we see inconsistent behavior with different dongles, even across dongles that appear to be the same make/model. If you have a spare microSD, the easiest thing would be to install Raspbian and then TinyPilot and do a quick test for H.264 mode.

closed 2023-03-03 by mtlynch

jlian, 2023-09-11:
Finally got the chance to do the update today and H.264 works quite well with the [HDMI dongle I have](https://www.amazon.com/gp/product/B09FLN63B3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). I saw that the USB init scripts moved places so I just had to reapply that but everything was straightforward.

mtlynch, 2023-09-15:
Oh, awesome! Thanks for the update!

You also might be interested to hear that there's a newer version of the HDMI to USB dongle that supports more advanced encoding options:

https://github.com/tiny-pilot/tinypilot/discussions/1593

I've got it on my shelf, but I haven't had time to experiment with it yet.