tiny-pilot/tinypilot #1351: Flag a warning in the debug logs if the customer is using unsupported hardware

jdeanwallace, 2023-04-05:
This issue has been recreated from https://github.com/tiny-pilot/ansible-role-tinypilot/issues/229

Sometimes, customers file support requests saying that TinyPilot doesn't work on their device, and it takes a few back-and-forths to discover that they're running it on unsupported hardware.

We should flag a warning in the debug logs if we detect that the running hardware does not match compatible hardware (Pi 4B or Pi Zero series). Only Pi 4B is officially supported, but Pi Zero should also at least work out of the box.

Depends on https://github.com/tiny-pilot/tinypilot/issues/1220

jdeanwallace, 2023-04-05:
@mtlynch - I recreated this issue from the now archived https://github.com/tiny-pilot/ansible-role-tinypilot/issues/229. Can we remove that issue from the project?

mtlynch, 2023-04-05:
@jdeanwallace - Sure, done!

cghague, 2024-07-02:
I spotted this issue while sorting through the Support Engineering task list, and I think we can close it without any action.

We already include the [hardware details](https://github.com/tiny-pilot/tinypilot/blob/6f37e68acbb729d82b76ed0dad0f67763926f29d/debian-pkg/opt/tinypilot-privileged/scripts/collect-debug-logs#L90) in the logs, and it's standard practice for us to check this information when providing support, so a direct warning seems unnecessary.

It's also worth noting that we included logic in the installer to catch the most common unsupported hardware and software that users tend to try, such as installing on [the Raspberry Pi 3](https://github.com/tiny-pilot/tinypilot/blob/6f37e68acbb729d82b76ed0dad0f67763926f29d/bundler/bundle/install#L14) or trying to use an [outdated](https://github.com/tiny-pilot/tinypilot/blob/6f37e68acbb729d82b76ed0dad0f67763926f29d/bundler/bundle/install#L21) or [not yet compatible](https://github.com/tiny-pilot/tinypilot/blob/6f37e68acbb729d82b76ed0dad0f67763926f29d/bundler/bundle/install#L33) version of Raspberry Pi OS.

@jdeanwallace - Do you agree this can be closed?

jdeanwallace, 2024-07-03:
@cghague - Agreed, we can close this.

closed 2024-07-03 by cghague