tiny-pilot/tinypilot #1365: Websocket / proxy / host is not an accepted origin.

dkerr64, 2023-04-19:
I have run into the issue documented in #658 and the workaround suggested there...
```
socketio = flask_socketio.SocketIO(cors_allowed_origins="*")
```
works.

But that feels like a hack.  What I have is a `lighttpd` server acting as a proxy, so I can access `tinypilot` from "outside" with certificate based authentication...

```
$SERVER["socket"] == "0.0.0.0:8082" {
    # tinyproxy access
    # require client certificate authentication
    # requires header upgrade/enable for Websockets support
    ssl.engine                  = "enable"
    ssl.cipher-list             = "aRSA+HIGH !3DES +kEDH +kRSA !kSRP !kPSK"
    ssl.pemfile                 = "/mnt/kd/ssl/webinterface.pem"
    ssl.ca-file                 = "/root/ca/intermediate/certs/ca-chain.cert.pem"
    ssl.verifyclient.activate   = "enable"
    ssl.verifyclient.enforce    = "enable"
    ssl.verifyclient.depth      = 2
    ssl.verifyclient.username   = "SSL_CLIENT_S_DN_CN"
    server.document-root        = ""
    proxy.server += ( "" =>
      ((
        "host" => "192.168.17.13",
        "port" => "80"
      ))
    )
    proxy.header = ( "upgrade" => "enable" )
}
```
but the websocket connection is failing, as documented in several other issues and #658

With `tcpdump` I have captured this...

```
18:00:08.138664 IP (tos 0x0, ttl 64, id 40890, offset 0, flags [DF], proto TCP (6), length 889)
    192.168.17.1.53576 > 192.168.17.13.80: Flags [P.], cksum 0xa6ca (incorrect -> 0x971e), seq 1:838, ack 1, win 502, options [nop,nop,TS val 45047007 ecr 643309545], length 837: HTTP, length: 837
	POST /socket.io/?EIO=4&transport=polling&t=OURY3Qp&sid=sJJijkijyZaBZMkxAABl HTTP/1.0
	Host: 192.168.17.1:8082
	Accept: */*
	Sec-Fetch-Site: same-origin
	Accept-Language: en-US,en;q=0.9
	Accept-Encoding: gzip, deflate, br
	Sec-Fetch-Mode: cors
	Content-Type: text/plain;charset=UTF-8
	Origin: https://192.168.17.1:8082
	User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.4 Safari/605.1.15
	Referer: https://192.168.17.1:8082/
	Content-Length: 1
	Sec-Fetch-Dest: empty
	Cookie: session=eyJjc3JmX3Rva2VuIjoiOWI3NmU0Y2UwYmMxNmRiNWQwNDlmMjYyMTk4NjE4NGY5Y2U0ZTM2NiJ9.ZEBkZQ.mV1whYd8oOnUEBubnT7oOwscu_0; stream_client=0/72603f3ce3ee4619
	X-Forwarded-For: 192.168.17.237
	X-Host: 192.168.17.1:8082
	X-Forwarded-Host: 192.168.17.1:8082
	X-Forwarded-Proto: https
	Connection: close

	1 [|http]
E..y..@.@..e.........H.P.....;.............
..\.&X..POST /socket.io/?EIO=4&transport=polling&t=OURY3Qp&sid=sJJijkijyZaBZMkxAABl HTTP/1.0
Host: 192.168.17.1:8082
Accept: */*
Sec-Fetch-Site: same-origin
Accept-Language: en-US,en;q=0.9
Accept-Encoding: gzip, deflate, br
Sec-Fetch-Mode: cors
Content-Type: text/plain;charset=UTF-8
Origin: https://192.168.17.1:8082
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.4 Safari/605.1.15
Referer: https://192.168.17.1:8082/
Content-Length: 1
Sec-Fetch-Dest: empty
Cookie: session=eyJjc3JmX3Rva2VuIjoiOWI3NmU0Y2UwYmMxNmRiNWQwNDlmMjYyMTk4NjE4NGY5Y2U0ZTM2NiJ9.ZEBkZQ.mV1whYd8oOnUEBubnT7oOwscu_0; stream_client=0/72603f3ce3ee4619
X-Forwarded-For: 192.168.17.237
X-Host: 192.168.17.1:8082
X-Forwarded-Host: 192.168.17.1:8082
X-Forwarded-Proto: https
Connection: close

1
18:00:08.139048 IP (tos 0x0, ttl 64, id 59309, offset 0, flags [DF], proto TCP (6), length 52)
    192.168.17.13.80 > 192.168.17.1.53576: Flags [.], cksum 0x728c (correct), seq 1, ack 838, win 503, options [nop,nop,TS val 643309545 ecr 45047007], length 0
E..4..@.@............P.H.;..........r......
&X....\.
18:00:08.141554 IP (tos 0x0, ttl 64, id 59310, offset 0, flags [DF], proto TCP (6), length 258)
    192.168.17.13.80 > 192.168.17.1.53576: Flags [P.], cksum 0x952b (correct), seq 1:207, ack 838, win 503, options [nop,nop,TS val 643309548 ecr 45047007], length 206: HTTP, length: 206
	HTTP/1.1 400 BAD REQUEST
	Server: nginx/1.14.2
	Date: Wed, 19 Apr 2023 22:00:08 GMT
	Content-Type: text/plain
	Content-Length: 54
	Connection: close

	"https://192.168.17.1:8082 is not an accepted origin." [|http]
E.....@.@............P.H.;...........+.....
&X....\.HTTP/1.1 400 BAD REQUEST
Server: nginx/1.14.2
Date: Wed, 19 Apr 2023 22:00:08 GMT
Content-Type: text/plain
Content-Length: 54
Connection: close

"https://192.168.17.1:8082 is not an accepted origin."
```
And it was that "not an accepted origin" message that allowed me to find #658 and try out the suggested fix... which works.

BUT, and this is my question... what could/should I do at the proxy to make this work?  If I could change or add anything to the HTTP headers, what should I change it to?  Over on the lighttpd site, any discussion of problems with proxy/websocket, they are adamant that they are following the specs and if there is a problem it is not with them, but is with the target host !!

Thanks!

dkerr64, 2023-04-20:
I have found a workaround that is possibly better than the suggestion in #658 (no code change)

The problem is, I think, that the origin and forwarded protocol headers state HTTPS when tinypilot expects (requires) HTTP.  So I decided to try and see what happens if I enable SSL on tinypilot following instructions [here](https://github.com/tiny-pilot/tinypilot/wiki/Enabling-SSL-with-TinyPilot).

I did that, and then because `lighttpd` can only proxy to HTTP, I needed to add in `stunnel` which turns the HTTP back into HTTPS.  Which looks like...
```
        https                   http             https
client -------> lighttpd proxy -------> stunnel -------> tinypilot
```
And this works.  The proxy and stunnel are on the same host, stunnel is listening on 127.0.0.1 and forwarding to tinypilot on port 443.

Now I suspect that if I could change the `Origin` and `X-Forwarded-Proto` headers in `lighttpd` to be http and not https then it would also work, but I haven't figured out how to do that yet.  But no harm adding SSL to Tinypilot, and I need to have stunnel running to support other https backends, so I'm satisfied with this workaround.

mtlynch, 2023-04-20:
Thanks for reporting this, @dkerr64.

I'm not familiar with lighthttpd, but from your description, it sounds like the issue is a configuration issue with lighthttpd.

> The problem is, I think, that the origin and forwarded protocol headers state HTTPS when tinypilot expects (requires) HTTP. So I decided to try and see what happens if I enable SSL on tinypilot [...]

This shouldn't be necessary. The client (browser) and lighttpd should have a secure TLS connection, but lighttpd should terminate the TLS connection and forward the HTTP request to TinyPilot via plaintext HTTP. The only reason to encrypt the lighttpd <-> TinyPilot connection is if it was going over an untrusted channel (e.g., the Internet), but it sounds like they're both running on localhost.

I think what you want is to configure your lighttpd to match [TinyPilot's nginx config](https://github.com/tiny-pilot/tinypilot/blob/43eb80c70755fe1af960c4903285a33bde69f08b/ansible-role/tasks/nginx.yml#L31L42). Two differences I noticed:

* TinyPilot's nginx config sets the Connection header to `Upgrade` whereas your lighttpd sets it to `close`
* TinyPilot's nginx config sets the HTTP protocol to 1.1, whereas your lighttpd sets it to 1.0

I'm not sure if these are tripping up socketio, but something about the headers is causing socketio to reject the origin even though socketio definitely supports HTTP proxies like lighttpd.

dkerr64, 2023-04-20:
Yes, I agree... I'll keep looking at `lighttpd` [config options](https://redmine.lighttpd.net/projects/lighttpd/wiki/Mod_proxy) and see if I can persuade it to set the headers to HTTP.  To be clear, the proxy is terminating the TLS connection and forwarded packets are not encrypted, but right now I don't understand why it is setting those two headers to https.

If I find out, I'll be sure to post here.

Thanks
David

mtlynch, 2023-04-21:
>right now I don't understand why it is setting those two headers to https.

Which headers?

`X-Forwarded-Proto: https` is expected because the browser connected to lighttpd over HTTPS, so lighttpd is accurately reporting that as it forwards the HTTP request. See: https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/X-Forwarded-Proto

gstrauss, 2023-04-24:
@dkerr64  try adding this to your lighttpd.conf reverse-proxy config:
`proxy.header += ("https-remap" => "enable")`

https://wiki.lighttpd.net/mod_proxy

dkerr64, 2023-04-25:
> @dkerr64 try adding this to your lighttpd.conf reverse-proxy config: `proxy.header += ("https-remap" => "enable")`
> 
> https://wiki.lighttpd.net/mod_proxy

I have tried that and it does not help.

dkerr64, 2023-04-25:
> Which headers?
>
> `X-Forwarded-Proto: https` is expected because the browser connected to lighttpd over HTTPS, so lighttpd is accurately reporting that as it forwards the HTTP request. See: [...]

What is expected in the `Origin` header?  It is being set to https://<ipaddr>:<port> should it be http?

gstrauss, 2023-04-26:
> > @dkerr64 try adding this to your lighttpd.conf reverse-proxy config: `proxy.header += ("https-remap" => "enable")`
> > https://wiki.lighttpd.net/mod_proxy
> 
> I have tried that and it does not help.

Did you try that literally, or did you look at the documentation?  As documented, that has no effect without associated "map-host-request" and/or "map-host-response".  I don't know the specific requirements of tiny-pilot, so I can not give you a complete example.  A naive attempt might be
```
proxy.header += (
    "https-remap" => "enable",
    "map-host-request"  => ("-" => "-"),
    "map-host-response" => ("-" => "-")
)
```
though the "DWIM" magic is not guaranteed.  Please refer to the documentation for further details:
https://wiki.lighttpd.net/mod_proxy

mtlynch, 2023-06-02:
Closing due to lack of response + seems to be an issue outside of TinyPilot.

closed 2023-06-02 by mtlynch