tiny-pilot/tinypilot #1372: Failed to download key at https://ftp-master.debian.org/keys/archive-key-n/a.asc

yaniv15, 2023-04-22:
## Description
I am trying to install tinypilot on Odroid C4 running 22.04 kernel 5.17, when performing a simple install I got this error
**TASK [ansible-role-ustreamer : add archive apt repository key] **********************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to download key at https://ftp-master.debian.org/keys/archive-key-n/a.asc: HTTP Error 404: Not Found"}**

Briefly summarize the problem you're experiencing.
The installation stops and doesn't move forward

## What's the behavior that you expect?
simple install

_Your answer_

## What's happening instead?

_Your answer_

## What are the steps to reproduce this behavior?
1. run the command
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  sudo reboot
2. get the error
TASK [ansible-role-ustreamer : add archive apt repository key] **********************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to download key at https://ftp-master.debian.org/keys/archive-key-n/a.asc: HTTP Error 404: Not Found"}

PLAY RECAP **************************************************************************************************************************
localhost                  : ok=9    changed=0    unreachable=0    failed=1    skipped=2    rescued=0    ignored=1
 

## Screenshots

_If applicable, add screenshots to help explain your problem._

## Logs
Collecting ansible==2.10.7
  Downloading ansible-2.10.7.tar.gz (29.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 29.9/29.9 MB 9.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting ansible-base==2.10.17
  Downloading ansible-base-2.10.17.tar.gz (6.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.1/6.1 MB 17.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting cffi==1.15.1
  Downloading cffi-1.15.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (470 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 470.6/470.6 kB 14.3 MB/s eta 0:00:00
Collecting cryptography==37.0.4
  Downloading cryptography-37.0.4-cp36-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.manylinux_2_24_aarch64.whl (3.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.7/3.7 MB 18.1 MB/s eta 0:00:00
Collecting Jinja2==3.1.2
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 8.3 MB/s eta 0:00:00
Collecting MarkupSafe==2.1.1
  Downloading MarkupSafe-2.1.1.tar.gz (18 kB)
  Preparing metadata (setup.py) ... done
Collecting packaging==21.3
  Downloading packaging-21.3-py3-none-any.whl (40 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 40.8/40.8 kB 1.5 MB/s eta 0:00:00
Collecting pycparser==2.21
  Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 118.7/118.7 kB 7.5 MB/s eta 0:00:00
Collecting pyparsing==3.0.9
  Downloading pyparsing-3.0.9-py3-none-any.whl (98 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 98.3/98.3 kB 6.3 MB/s eta 0:00:00
Collecting PyYAML==6.0
  Downloading PyYAML-6.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (732 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 732.4/732.4 kB 13.8 MB/s eta 0:00:00
Installing collected packages: PyYAML, pyparsing, pycparser, MarkupSafe, packaging, Jinja2, cffi, cryptography, ansible-base, ansible
  DEPRECATION: MarkupSafe is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for MarkupSafe ... done
  DEPRECATION: ansible-base is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for ansible-base ... done
  DEPRECATION: ansible is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for ansible ... done
Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 PyYAML-6.0 ansible-2.10.7 ansible-base-2.10.17 cffi-1.15.1 cryptography-37.0.4 packaging-21.3 pycparser-2.21 pyparsing-3.0.9
+ ansible-playbook --inventory localhost, install.yml --extra-vars @/var/tmp/tmp.yXYLTKXVLR.yml --extra-vars tinypilot_debian_package_path=/opt/tinypilot-updater/tinypilot_20230419124618_all.deb

PLAY [localhost] ********************************************************************************************************************

TASK [Gathering Facts] **************************************************************************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : include distribution-specific vars] ******************************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : check that the H264 variables are in a consistent state] *********************************************
skipping: [localhost]

TASK [ansible-role-ustreamer : determine whether to install Janus] ******************************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : detect OS] *******************************************************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : decide whether to support audio streaming] ***********************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : install Janus] ***************************************************************************************
included: /opt/tinypilot-updater/roles/ansible-role-ustreamer/tasks/install_janus.yml for localhost

TASK [ansible-role-ustreamer : check if legacy Janus package is installed] **********************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "dpkg-query --show --showformat='${Maintainer}' janus | grep TinyPilot", "delta": "0:00:00.027244", "end": "2023-04-22 05:36:50.975450", "msg": "non-zero return code", "rc": 1, "start": "2023-04-22 05:36:50.948206", "stderr": "dpkg-query: no packages found matching janus", "stderr_lines": ["dpkg-query: no packages found matching janus"], "stdout": "", "stdout_lines": []}
...ignoring

TASK [ansible-role-ustreamer : determine if legacy Janus package is installed] ******************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : uninstall legacy Janus package] **********************************************************************
skipping: [localhost]

TASK [ansible-role-ustreamer : install apt_key dependency] **************************************************************************
ok: [localhost]

TASK [ansible-role-ustreamer : add archive apt repository key] **********************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to download key at https://ftp-master.debian.org/keys/archive-key-n/a.asc: HTTP Error 404: Not Found"}

PLAY RECAP **************************************************************************************************************************
localhost                  : ok=9    changed=0    unreachable=0    failed=1    skipped=2    rescued=0    ignored=1

+ clean_up
+ rm -rf /var/tmp/tmp.yXYLTKXVLR.yml
+ clean_up
+ rm -rf /var/tmp/tmp.jhCzEJjfuO.tgz

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

yaniv15, 2023-04-22:
I don't know the specific reason but the variable is tasks are not being initialized.
for example the above issue is caused by {{ ansible_distribution_major_version }}
I tried to bypass this, but then struck with {{ ustreamer_janus_apt_suite }}.
Then II realized none of the variables present in the tasks is being initialized.
what could be the possible issue?
Where are these variables initialized?
can we manually initialze these variables?

mtlynch, 2023-04-24:
>I am trying to install tinypilot on Odroid C4 running 22.04 kernel 5.17, when performing a simple install I got this error

Unfortunately, we only provide support for Raspberry Pi systems.

I can try to point you in the right direction in the [alternate hardware](https://github.com/tiny-pilot/tinypilot/discussions/755#discussioncomment-5692215) thread.

closed 2023-04-24 by mtlynch