tiny-pilot/tinypilot #1392: Fresh install on PI Zero 2W Fails (PI O/S Lite 32 Bit Bullseye)

pacmac, 2023-05-14:
Fails every time same place.

There's nothing that I can see that would help me to debug the cause:

```
root@tinypilot:/home/pi# curl   --silent   --show-error   https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh |     bash - &&   reboot
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ [[ . = \. ]]
+ SCRIPT_DIR=/opt/tinypilot
+ readonly SCRIPT_DIR
+ readonly TINYPILOT_README=/opt/tinypilot/README.md
+ TINYPILOT_README=/opt/tinypilot/README.md
+ [[ -f /opt/tinypilot/README.md ]]
+ readonly HAS_PRO_INSTALLED
+ [[ 0 = 1 ]]
+ readonly LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ readonly INSTALLER_DIR=/mnt/tinypilot-installer
+ INSTALLER_DIR=/mnt/tinypilot-installer
+ readonly BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
+ BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
+ readonly RAMDISK_SIZE=500m
+ RAMDISK_SIZE=500m
+ trap clean_up EXIT
+ sudo mkdir /mnt/tinypilot-installer
+ sudo mount --types tmpfs --options size=500m --source tmpfs --target /mnt/tinypilot-installer --verbose
mount: tmpfs mounted on /mnt/tinypilot-installer.
++ curl https://gk.tinypilotkvm.com/community/download/latest --location --output /mnt/tinypilot-installer/bundle.tgz --write-out '%{http_code}' --silent
+ HTTP_CODE=200
+ readonly HTTP_CODE
+ [[ 200 != \2\0\0 ]]
+ sudo tar --gunzip --extract --file /mnt/tinypilot-installer/bundle.tgz --directory /mnt/tinypilot-installer
+ sudo chown root:root --recursive /mnt/tinypilot-installer
+ [[ 0 -eq 1 ]]
+ pushd /mnt/tinypilot-installer
/mnt/tinypilot-installer /home/pi
+ sudo ./install
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
+ . lib.sh
+ readonly TEMP_DIR=/var/tmp
+ TEMP_DIR=/var/tmp
++ mktemp --tmpdir=/var/tmp --suffix .yml
+ INSTALL_SETTINGS_FILE=/var/tmp/tmp.3Y3LqCOyz4.yml
+ readonly INSTALL_SETTINGS_FILE
+ readonly TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ readonly USTREAMER_SETTINGS_FILE=/home/ustreamer/config.yml
+ USTREAMER_SETTINGS_FILE=/home/ustreamer/config.yml
++ ls tinypilot_20230511144202_all.deb
+ TINYPILOT_DEBIAN_PACKAGE=tinypilot_20230511144202_all.deb
+ readonly TINYPILOT_DEBIAN_PACKAGE
+ trap clean_up EXIT
+ grep -q '^Model *: Raspberry Pi 3' /proc/cpuinfo
++ lsb_release --id --short
+ [[ Raspbian == \R\a\s\p\b\i\a\n ]]
++ lsb_release --release --short
+ ((  11 < 11  ))
+ grep -q boot=overlay /proc/cmdline
+ [[ -f /home/tinypilot/settings.yml ]]
+ echo 'Using settings file at: /home/tinypilot/settings.yml'
Using settings file at: /home/tinypilot/settings.yml
+ cp /home/tinypilot/settings.yml /var/tmp/tmp.3Y3LqCOyz4.yml
+ yaml_set_if_undefined /var/tmp/tmp.3Y3LqCOyz4.yml ustreamer_port 8001
+ local file_path=/var/tmp/tmp.3Y3LqCOyz4.yml
+ local key=ustreamer_port
+ local value=8001
+ grep --silent '^ustreamer_port:' /var/tmp/tmp.3Y3LqCOyz4.yml
+ echo 'ustreamer_port: 8001'
+ tee --append /var/tmp/tmp.3Y3LqCOyz4.yml
ustreamer_port: 8001
+ yaml_set_if_undefined /var/tmp/tmp.3Y3LqCOyz4.yml ustreamer_persistent true
+ local file_path=/var/tmp/tmp.3Y3LqCOyz4.yml
+ local key=ustreamer_persistent
+ local value=true
+ grep --silent '^ustreamer_persistent:' /var/tmp/tmp.3Y3LqCOyz4.yml
+ echo 'ustreamer_persistent: true'
+ tee --append /var/tmp/tmp.3Y3LqCOyz4.yml
ustreamer_persistent: true
+ USE_TC358743_DEFAULTS=false
+ grep --silent '^ustreamer_capture_device:' /var/tmp/tmp.3Y3LqCOyz4.yml
+ grep --silent '^ustreamer_capture_device: tc358743$' /var/tmp/tmp.3Y3LqCOyz4.yml
+ USE_TC358743_DEFAULTS=true
+ true
+ echo 'Final install settings:'
Final install settings:
+ cat /var/tmp/tmp.3Y3LqCOyz4.yml
ustreamer_capture_device: tc358743
ustreamer_port: 8001
ustreamer_persistent: true
+ apt-get update --allow-releaseinfo-change-suite
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
+ apt-get install -y git libffi-dev libssl-dev python3-dev python3-venv lsb-release
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1+deb11u2).
libffi-dev is already the newest version (3.3-6).
lsb-release is already the newest version (11.1.0+rpi1).
python3-dev is already the newest version (3.9.2-3).
python3-venv is already the newest version (3.9.2-3).
libssl-dev is already the newest version (1.1.1n-0+deb11u4+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ python3 -m venv venv
+ . venv/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/mnt/tinypilot-installer/venv
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
++ PATH=/mnt/tinypilot-installer/venv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ PS1='(venv) '
++ export PS1
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
+ pip install 'pip>=21.3.1'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip>=21.3.1
  Using cached https://www.piwheels.org/simple/pip/pip-23.1.2-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-23.1.2
+ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible==2.10.7 (from -r requirements.txt (line 5))
  Using cached https://www.piwheels.org/simple/ansible/ansible-2.10.7-py3-none-any.whl (48.2 MB)
Collecting ansible-base==2.10.17 (from -r requirements.txt (line 8))
  Using cached https://www.piwheels.org/simple/ansible-base/ansible_base-2.10.17-py3-none-any.whl (1.9 MB)
Collecting cffi==1.15.1 (from -r requirements.txt (line 9))
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting cryptography==37.0.4 (from -r requirements.txt (line 10))
  Using cached https://www.piwheels.org/simple/cryptography/cryptography-37.0.4-cp39-cp39-linux_armv7l.whl (2.5 MB)
Collecting Jinja2==3.1.2 (from -r requirements.txt (line 11))
  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting MarkupSafe==2.1.1 (from -r requirements.txt (line 12))
  Using cached https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.1-cp39-cp39-linux_armv7l.whl (23 kB)
Collecting packaging==21.3 (from -r requirements.txt (line 13))
  Using cached https://www.piwheels.org/simple/packaging/packaging-21.3-py3-none-any.whl (40 kB)
Collecting pycparser==2.21 (from -r requirements.txt (line 14))
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Collecting pyparsing==3.0.9 (from -r requirements.txt (line 15))
  Using cached https://www.piwheels.org/simple/pyparsing/pyparsing-3.0.9-py3-none-any.whl (98 kB)
Collecting PyYAML==6.0 (from -r requirements.txt (line 16))
  Using cached https://www.piwheels.org/simple/pyyaml/PyYAML-6.0-cp39-cp39-linux_armv7l.whl (45 kB)
Installing collected packages: PyYAML, pyparsing, pycparser, MarkupSafe, packaging, Jinja2, cffi, cryptography, ansible-base, ansible
./install: line 126:   826 Killed                  pip install -r requirements.txt
+ clean_up
+ rm -rf /var/tmp/tmp.3Y3LqCOyz4.yml
+ clean_up
+ umount --lazy /mnt/tinypilot-installer
+ rm -rf /opt/tinypilot-updater /mnt/tinypilot-installer
```

pacmac, 2023-05-14:
If I download the script and prevent removal on failure and modify the install file by commenting out set -e

```
# Exit on first error.
#set -e
```

I get:

```
TASK [ansible-role-ustreamer : add archive apt repository key] ********************************************************************************************************************
changed: [localhost]

TASK [ansible-role-ustreamer : enable Janus apt suite] ****************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "module_stderr": "Killed\n", "module_stdout": "", "msg": "MODULE FAILURE\nSee stdout/stderr for the exact error", "rc": 137}

PLAY RECAP ************************************************************************************************************************************************************************
localhost                  : ok=8    changed=1    unreachable=0    failed=1    skipped=1    rescued=0    ignored=0
```

pacmac, 2023-05-14:
The solution was to create a 1GB swap file.

pacmac, 2023-05-14:
I see that Janus is using 10 threads, each one is using 4% of memory.

is it possible to optimise this to reduce the memory footprint ?

[image]

mtlynch, 2023-05-15:
Thanks for reporting this!

>and journalctl shows:

>Out of memory: Killed process 1881 (python3) total-vm:64276kB, anon-rss:34512kB, file-rss:8kB, shmem-rss:0kB, UID:0 pgtables:60kB oom_score_adj:0

Ah, this makes sense. To minimize disk wear, we recently switched to performing installs in a 500 MB RAMdisk (#1360). On a Pi 4, it's fine, but on a Pi Zero 2 W, that consumes all the available memory.

I'm not sure why you're seeing so many Janus processes on the Zero. On a Pi 4, I'm only seeing one:

[image]

pacmac, 2023-05-15:
I see 12 processes on PI4, see attached.
[image]

pacmac, 2023-05-16:
This is PI Zero, there does seem to be a lot of threads running:

[image]

pacmac, 2023-05-17:
I wont continue with this issue, but just to let you know that I am unable to update:
```
ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device: '/mnt/tinypilot-installer/venv/lib/python3.9/site-packages/ansible-2.10.7.dist-info/INSTALLERnirgnnl4.tmp'
```

There's plenty of free space on the disk, so I guess this is ram-disk related.

Would it be possible to include a option not to use ram disk for installation ??

mtlynch, 2023-05-17:
>I see 12 processes on PI4, see attached.

Ah, I see. If I check htop, I see the same.

>There's plenty of free space on the disk, so I guess this is ram-disk related.

>Would it be possible to include a option not to use ram disk for installation ??

Oh, that's surprising. I'd expect it to fail to create the RAMdisk if it's out of RAM, but it has enough RAM to create the RAMdisk but it's running out of space. Is this on the Pi Zero 2 W? Can you share the full logs in a gist?

pacmac, 2023-05-17:
https://gist.githubusercontent.com/pacmac/4b88b4a8609466976fbeb250d2ddb741/raw/8e869eb1c7c65e7d0ee989b23a47dc6a8f456576/tinypilot_update_230518

pacmac, 2023-05-17:
OK, if I change the TEMP_DIR in the install file:

```
readonly TEMP_DIR=/root/tmp

```

and then
```
cd /mnt/tinypilot-installer/
./install
```

no errors and the update / install is sucessful.

mtlynch, 2023-05-18:
@pacmac - Ah, that's helpful. Thanks!

I've filed #1402 to see if we can fix this so it goes back to "just works" on the Pi Zero.

mtlynch, 2023-05-26:
@pacmac - This should be working as of #1406. If you have a moment to test the install on your Pi Zero 2 W, it'd be helpful to confirm the fix.

pacmac, 2023-05-27:
Thanks, yep a fresh installation works, I still had to reduce the GPU Mam down to 128MB as the default of 256 uses half of the available memory.

Not sure if the update will work, will only find out when there's another update.

mtlynch, 2023-05-29:
@pacmac - Cool, thanks! The free version doesn't do versioned updates, so an update is available anytime there's a new commit to master, so you should be able to update whenever you're ready.

pacmac, 2023-06-01:
Thanks, I just attempted an update from the web console, unfortunately it failed: 

https://logs.tinypilotkvm.com/2qDyuYlI

mtlynch, 2023-06-01:
@pacmac - I think the issue there was something to do with clock skew on your device, so I don't think it was a RAM problem. 

The clock at the start of the update was `2023-05-31 09:05:39.268`, but then at the end it was `2023-06-01 17:31:28.344`. It didn't really run for 36+ hours, right?

```
2023-05-31 09:05:39.268 __main__        INFO Starting TinyPilot update
2023-05-31 09:05:39.268 __main__        INFO Launching update script: /opt/tinypilot-privileged/scripts/update
...
tar: ./ansible.cfg: time stamp 2023-05-31 20:54:17 is 42513.279881804 s in the future
tar: ./install: time stamp 2023-05-31 20:54:17 is 42513.27934196 s in the future
...
2023-06-01 17:31:28.344 __main__        ERROR Update process timed out
```

Did you run the TinyPilot update soon after the Zero started up? Did the Zero not have Internet access until the update was underway?

What I suspect happened was that the Zero was turned off so its internal clock was behind. It contacted a time server in the middle of the update, which caused the time to jump ahead to the next day. Then the update in the system clock caused the update process to think TinyPilot hung during the update.

pacmac, 2023-06-01:
singapore 11:30pm

pacmac, 2023-06-02:
that update killed the frontend, so I ran the updated from backend.

If I recall correctly, the process dying the way this did was due to out of memory.

```
root@tinypilot:~# date
Fri Jun  2 07:37:31 +08 2023
root@tinypilot:~#
root@tinypilot:~#
root@tinypilot:~#
root@tinypilot:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:4c:77:ee brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.234/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 6190sec preferred_lft 5409sec
    inet6 fe80::e765:2caf:e7ca:a400/64 scope link
       valid_lft forever preferred_lft forever
root@tinypilot:~#
root@tinypilot:~#
root@tinypilot:~#
root@tinypilot:~# curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  sudo reboot
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ [[ . = \. ]]
+ SCRIPT_DIR=/opt/tinypilot
+ readonly SCRIPT_DIR
+ readonly TINYPILOT_README=/opt/tinypilot/README.md
+ TINYPILOT_README=/opt/tinypilot/README.md
+ [[ -f /opt/tinypilot/README.md ]]
++ head -n 1 /opt/tinypilot/README.md
+ [[ # TinyPilot = \#\ \T\i\n\y\P\i\l\o\t\ \P\r\o ]]
+ readonly HAS_PRO_INSTALLED
+ [[ 0 = 1 ]]
+ readonly LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ readonly RAMDISK_SIZE_MIB=500
+ RAMDISK_SIZE_MIB=500
++ free --mebi
++ grep --fixed-strings Mem:
++ tr --squeeze-repeats ' '
++ cut --delimiter ' ' --fields 4
+ FREE_MEMORY_MIB=91
+ readonly FREE_MEMORY_MIB
+ INSTALLER_DIR=/mnt/tinypilot-installer
+ trap clean_up EXIT
+ ((  91 >= 500  ))
++ mktemp --directory
+ INSTALLER_DIR=/tmp/tmp.070Rj0Qxso
+ readonly INSTALLER_DIR
+ readonly BUNDLE_FILE=/tmp/tmp.070Rj0Qxso/bundle.tgz
+ BUNDLE_FILE=/tmp/tmp.070Rj0Qxso/bundle.tgz
++ curl https://gk.tinypilotkvm.com/community/download/latest --location --output /tmp/tmp.070Rj0Qxso/bundle.tgz --write-out '%{http_code}' --silent
+ HTTP_CODE=200
+ readonly HTTP_CODE
+ [[ 200 != \2\0\0 ]]
+ sudo tar --gunzip --extract --file /tmp/tmp.070Rj0Qxso/bundle.tgz --directory /tmp/tmp.070Rj0Qxso
+ sudo chown root:root --recursive /tmp/tmp.070Rj0Qxso
[... 89 lines cut ...]
++ hash -r
+ pip install 'pip>=21.3.1'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip>=21.3.1
  Using cached https://www.piwheels.org/simple/pip/pip-23.1.2-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-23.1.2
+ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible==2.10.7 (from -r requirements.txt (line 5))
./install: line 126:  2232 Killed                  pip install -r requirements.txt
+ clean_up
+ rm -rf /var/tmp/tmp.PHdUMFQ0FG.yml
+ clean_up
+ umount --lazy /tmp/tmp.070Rj0Qxso
umount: /tmp/tmp.070Rj0Qxso: not mounted.
+ true
+ rm -rf /opt/tinypilot-updater /tmp/tmp.070Rj0Qxso
root@tinypilot:~# date
Fri Jun  2 08:05:35 +08 2023
```

mtlynch, 2023-06-02:
It looks like it's running with only 91 MB of RAM at the start of the install process:

```text
++ free --mebi
++ grep --fixed-strings Mem:
++ tr --squeeze-repeats ' '
++ cut --delimiter ' ' --fields 4
+ FREE_MEMORY_MIB=91
```

We can try to make it work on a Zero with ~400 MB free, but trying to support down to 91 MB feels like we're getting too far outside of the officially supported 4B with >= 1 GB of RAM.

pacmac, 2023-06-02:
That's because the installer assigns half the memory to the GPU:

```
# cat /boot/config.txt | grep gpu_mem
gpu_mem=256

root@tinypilot:~# free --mebi
               total        used        free      shared  buff/cache   available
Mem:             236          66          90           0          80         120
Swap:             99          14          85
```

```
root@tinypilot:~# cat /boot/config.txt | grep gpu_mem
gpu_mem=128
root@tinypilot:~# free --mebi
               total        used        free      shared  buff/cache   available
Mem:             363          78         168           0         116         233
Swap:             99           0          99
```

I did make a note that I lowered mine in the version I hacked to 128, but I just did a standard install now, and did not reduce it.

I'm not sure why it would need so much GPU ram, and during installation, this could probably reduced down to 16 or 32

pacmac, 2023-06-02:
it's just the install process that is sucking up the memory, I have had it running for weeks and there's no memory issues at all when it's running, especially with GPU=128.

If I stop the nginx, ustreamer, janus and modemmaneger service before install then I can free up more memory, could also stop other services, and reduce GPU memory for install and of course revert them after install.

```
# free --mebi
               total        used        free      shared  buff/cache   available
Mem:             363          65         180           0         117         246
Swap:             99           0          99
```

pacmac, 2023-06-03:
I can run the installer on a system that has already installed by doing this:

```
sed -i '/^gpu_mem=/c\gpu_mem=128' /boot/config.txt
reboot
systemctl stop janus ustreamer nginx ModemManager tinypilot
free --mebi

curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  reboot
```

And it completes without error, unlike before.

mtlynch, 2023-06-05:
Ah, right. It's possible we could get away with allocating less RAM to GPU. We're generally not RAM-bound so we haven't experimented much with it.

The logic for setting 256 MB is currently set [in our Ansible logic](https://github.com/tiny-pilot/ansible-role-ustreamer/blob/02f37b869c8c585517a9c7b6cd95cfb7661f1490/tasks/provision_tc358743.yml#L17-L23), which we're in the process of [reimplementing as simpler bash scripts](https://github.com/tiny-pilot/tinypilot/issues/1353). Once we're setting it from a bash script, we could try checking total RAM and choosing 128M if total RAM is < 1GB.

pacmac, 2023-07-14:
This seems to be working now and I succeeded in doing an update which I could not do before.

It would be good though if the install script does not over-write my GPU ram settings with 256M as that is half the memory.

I also have one other issue, and dont know if it's ZERO specific, but sometimes the video does not display, and I have to go into the video settings, change something (i.e. framerate) and save it and it all comes alive again.

mtlynch, 2023-07-18:
>It would be good though if the install script does not over-write my GPU ram settings with 256M as that is half the memory.

It's unfortunately not something we're able to support right now, but we'll keep this in mind when we reimplement this functionality without Ansible.

>I also have one other issue, and dont know if it's ZERO specific, but sometimes the video does not display, and I have to go into the video settings, change something (i.e. framerate) and save it and it all comes alive again.

If you see it happen again, can you try capturing the logs before changing the video settings and opening a new bug?

I'm going to close this one out as the original issue is resolved, and I don't want to mix in other Zero-related issues to make it hard to keep track.

closed 2023-07-18 by mtlynch

pacmac, 2023-11-25:
Can you re-visit this again, every time I do an update, I have to go into the confiig file and change the 256m back to what I had it set originally.

As this is an update and not a clean install, surely it should attempt to re-use the existing configurations ?

mtlynch, 2023-11-27:
@pacmac - Can you file a separate bug that explains the behavior you expect to see?

To set expectations, we'll track the feature request, but our priority is supported configurations.