tiny-pilot/tinypilot #1438: Raspbian 64 bit Raspberry Pi 4 error

detectingdragon77, 2023-06-12:
## Description

Briefly summarize the problem you're experiencing.

## What's the behavior that you expect?

_Your answer_
A successful installation.
## What's happening instead?
I am receiving an error when trying to install tiny pilot, check uploaded log.
_Your answer_

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on '....'
3. See error

## Screenshots

_If applicable, add screenshots to help explain your problem._

## Logs

Please paste the URL you see when you run   @@`/opt/tinypilot/dev-scripts/dump-logs`
[log.txt](https://github.com/tiny-pilot/tinypilot/files/11727588/log.txt)

mtlynch, 2023-06-12:
Thanks for reporting this, @detectingdragon77.

Does this happen on a stock install of Raspbian? In the logs, it looks like you've installed lots of other packages, so your Debian packages might not be in a consistent state before the install.

detectingdragon77, 2023-06-12:
> Thanks for reporting this, @detectingdragon77.
> 
> Does this happen on a stock install of Raspbian? In the logs, it looks like you've installed lots of other packages, so your Debian packages might not be in a consistent state before the install.
> 
> Which OS version is this?

Raspberry Pi OS with desktop
Release date: May 3rd 2023
System: 64-bit
Kernel version: 6.1
Debian version: 11 (bullseye)
I ran apt upgrade before all this and rebooted.

mtlynch, 2023-06-13:
@detectingdragon77 - Thanks for the extra details!

@db39 - Can you try reproducing this?

db39, 2023-06-14:
@mtlynch - I've been able to reproduce this error.

First, I installed Raspberry Pi OS (64-BIT) (with desktop) using the Raspberry Pi imager.

Then I logged in via `ssh` and ran the TinyPilot install command:

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  sudo reboot
```

The initial issues were dependencies and permissions:
 
```text
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ [[ . = \. ]]
+ SCRIPT_DIR=/opt/tinypilot
+ readonly SCRIPT_DIR
+ readonly TINYPILOT_README=/opt/tinypilot/README.md
+ TINYPILOT_README=/opt/tinypilot/README.md
+ [[ -f /opt/tinypilot/README.md ]]
+ readonly HAS_PRO_INSTALLED
+ [[ 0 = 1 ]]
+ readonly LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ readonly RAMDISK_SIZE_MIB=500
+ RAMDISK_SIZE_MIB=500
++ free --mebi
++ grep --fixed-strings Mem:
++ tr --squeeze-repeats ' '
++ cut --delimiter ' ' --fields 4
+ FREE_MEMORY_MIB=383
+ readonly FREE_MEMORY_MIB
+ INSTALLER_DIR=/mnt/tinypilot-installer
+ trap clean_up EXIT
+ ((  383 >= 500  ))
++ mktemp --directory
+ INSTALLER_DIR=/tmp/tmp.gtUZaPnxWy
+ readonly INSTALLER_DIR
+ readonly BUNDLE_FILE=/tmp/tmp.gtUZaPnxWy/bundle.tgz
+ BUNDLE_FILE=/tmp/tmp.gtUZaPnxWy/bundle.tgz
++ curl https://gk.tinypilotkvm.com/community/download/latest --location --output /tmp/tmp.gtUZaPnxWy/bundle.tgz --write-out '%{http_code}' --silent
+ HTTP_CODE=200
+ readonly HTTP_CODE
+ [[ 200 != \2\0\0 ]]
+ sudo tar --gunzip --extract --file /tmp/tmp.gtUZaPnxWy/bundle.tgz --directory /tmp/tmp.gtUZaPnxWy
+ sudo chown root:root --recursive /tmp/tmp.gtUZaPnxWy
+ [[ 0 -eq 1 ]]
+ pushd /tmp/tmp.gtUZaPnxWy
/tmp/tmp.gtUZaPnxWy ~
+ sudo ./install
+ . lib.sh
+ readonly TEMP_DIR=/var/tmp
+ TEMP_DIR=/var/tmp
++ mktemp --tmpdir=/var/tmp --suffix .yml
+ INSTALL_SETTINGS_FILE=/var/tmp/tmp.Jknoaq0rqa.yml
+ readonly INSTALL_SETTINGS_FILE
+ readonly TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ readonly USTREAMER_SETTINGS_FILE=/home/ustreamer/config.yml
+ USTREAMER_SETTINGS_FILE=/home/ustreamer/config.yml
++ ls tinypilot_20230613161050_armhf.deb
+ TINYPILOT_DEBIAN_PACKAGE=tinypilot_20230613161050_armhf.deb
+ readonly TINYPILOT_DEBIAN_PACKAGE
[... 385 lines cut ...]
Without sudo, you may not be able to gain administrative privileges.\r
\r
If you would prefer to access the root account with su(1)\r
or by logging in directly,\r
you must set a root password with \"sudo passwd\".\r
\r
If you have arranged other means to access the root account,\r
and you are sure this is what you want,\r
you may bypass this check by setting an environment variable \r
(export SUDO_FORCE_REMOVE=yes).\r
\r
Refusing to remove sudo.\r
dpkg: error processing package sudo (--remove):\r
 installed sudo package pre-removal script subprocess returned error exit status 1\r
dpkg: too many errors, stopping\r
Errors were encountered while processing:\r
 sudo\r
Processing was halted because there were too many errors.\r
", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following packages were automatically installed and are no longer required:", "  blt cups-pk-helper docutils-common edid-decode flashrom fonts-piboto", "  freeglut3 giblib1 gir1.2-atk-1.0 gir1.2-gdkpixbuf-2.0 gir1.2-gmenu-3.0", "  gir1.2-gtk-3.0 gir1.2-harfbuzz-0.0 gir1.2-notify-0.7", "  gir1.2-packagekitglib-1.0 gir1.2-pango-1.0 gir1.2-polkit-1.0 gir1.2-secret-1", "  gnome-menus gtk2-engines-clearlookspix gtk2-engines-pixflat gui-pkinst", "  hplip-data i2c-tools ibverbs-providers javascript-common libaio1", "  libboost-iostreams1.74.0 libboost-thread1.74.0 libcupsimage2 libdaxctl1", "  libexpat1-dev libftdi1-2 libfuse2 libgdata-common libgdata22 libgfapi0", "  libgfrpc0 libgfxdr0 libglu1-mesa libglusterfs0 libgnome-menu-3-0", "  libgoa-1.0-0b libgoa-1.0-common libhpmud0 libi2c0 libibverbs1 libimagequant0", "  libjs-jquery libjs-sphinxdoc libjs-underscore libldb2 libmikmod3 libndctl6", "  libnma-common libnma0 libpmem1 libpmemblk1 libportmidi0 libpython3-dev", "  libpython3-stdlib libpython3.9-dev libqt5designer5 libqt5help5 libqt5test5", "  librados2 librbd1 librdmacm1 librtimulib-dev librtimulib-utils librtimulib7", "  libsane-hpaio libsdl-mixer1.2 libsdl-ttf2.0-0 libtalloc2 libtcl8.6", "  libtevent0 libtk8.6 libturbojpeg0 libwbclient0 lxde-common lxplug-bluetooth", "  lxplug-cputemp lxplug-ejecter lxplug-magnifier lxplug-menu lxplug-netman", "  lxplug-network lxplug-ptbatt lxplug-updater lxplug-volumepulse pi-greeter", "  pi-printer-support pishutdown pixflat-icons printer-driver-escpr", "  printer-driver-hpcups printer-driver-postscript-hp python-apt-common", "  python3-asttokens python3-blinker python3-bs4 python3-click python3-colorama", "  python3-debconf python3-docutils python3-html5lib python3-isort", "  python3-itsdangerous python3-jedi python3-jwt python3-logilab-common", "  python3-mccabe python3-mypy-extensions python3-opengl python3-parso", "  python3-pexpect python3-piexif python3-ptyprocess python3-pygments", "  python3-pyinotify python3-responses python3-roman python3-send2trash", "  python3-soupsieve python3-typing-extensions python3-v4l2", "  python3-webencodings python3-werkzeug read-edid rp-bookshelf rpd-plym-splash", "  rpd-wallpaper rpi.gpio-common scrot sgml-base tk8.6-blt2.5 xml-core xsel", "Use 'sudo apt autoremove' to remove them.", "The follow [... cut ...] ors, stopping", "Errors were encountered while processing:", " sudo", "Processing was halted because there were too many errors."]}

PLAY RECAP ********************************************************************************************************************************************************************************************************
localhost                  : ok=45   changed=0    unreachable=0    failed=1    skipped=20   rescued=0    ignored=0   
```

I'll continue looking into this.

detectingdragon77, 2023-06-14:
So I have an update, I figured out that the issue was not when building but when installing the compiled deb. It is compiling for armhf for some reason instead of arm64, and this caused conflicting dependencies between python3:arm64 and python3:armhf

mtlynch, 2023-06-14:
Ah, right. I forgot that as of #1352, our Debian packages are architecture specific.

64-bit was never officially supported, but for certain stretches, it just worked anyway. But right now, it's definitely not going to work as we're only building armhf (32-bit), and we don't have resources to support 64-bit.

Sorry, I understand it's a frustrating answer, but we're a small team so we can only support one architecture at the moment.

closed 2023-06-14 by mtlynch

detectingdragon77, 2023-06-14:
> Ah, right. I forgot that as of #1352, our Debian packages are architecture specific.
>
> 64-bit was never officially supported, but for certain stretches, it just worked anyway. But right now, it's [...]

Ok understood, thanks for the explanation.

spoelstraethan, 2023-07-22:
I think many distributions are moving away from 32 bit, and many armhf boards are very long in the tooth at this point, it shouldn't be a huge challenge to produce both 32 and 64 bit Debian packages because you are just relying on Python for most of the heavy lifting, and it handles the 32/64 bit translations for you, you just need to output both a 32bit armhf and a 64 bit arm64 debian. 

Check out this tool that can help make the process WAY simpler.

https://github.com/jordansissel/fpm

CHN-STUDENT, 2023-11-02:
@detectingdragon77 hi, could you write this suggestion to `readme.md`, today i try to install, but i meet same issue. I think you can give some suggestion when install!

detectingdragon77, 2023-11-02:
> @detectingdragon77 hi, could you write this suggestion to `readme.md`, today i try to install, but i meet same issue. I think you can give some suggestion when install!
> 
> Update: sorry, you give a suggestion we have to use 32bit OS, could you can make it in bold ?

I'm sorry, I've pretty much abandoned trying to use this.

CHN-STUDENT, 2023-11-03:
I am sorry because i try to suggest this project collaborator to add more readme. By the way. which the best way you use kvm ?