tiny-pilot/tinypilot #1460: Add support for specifying a STUN server or 1:1 mapping for Janus

mtlynch, 2023-06-21:
There are a few common scenarios where the browser and the Janus server on the TinyPilot device can't establish a WebRTC connection. See @cghague's investigations for details (especially the Conclusions note at the end):

https://github.com/tiny-pilot/tinypilotkvm.com/issues/939#issuecomment-1600620310

We should add an "Advanced" section to the Video Settings page that allows the user to configure a STUN server or 1:1 mapping for Janus to use in the WebRTC connection. For privacy, I'd like the STUN server to be off by default, but if they enable it, we should default to a Google STUN server but allow the user to enter any STUN server they choose.

It would be a nice bonus if we can detect whether the user is able to establish a WebRTC connection without a STUN server or 1:1 mapping and give them clearer guidance, but it's not critical that we do that.

mtlynch, 2023-07-13:
>Where do we maintain state and configuration?
>
>    (a.1) ~/settings.yml
>    (a.2) sqlite DB (like the video streaming mode setting)
>    (a.3) In the Janus config file directly (by grepping/parsing it)

Sidenote: We should come up with more general rules about what goes in `settings.yml` vs. what goes in SQLite.

>In regards to (a), I think (a.1) would be the easiest and most consistent place – we also still store all uStreamer parameters there. (a.2) would make it difficult to process the data outside of the Python app (e.g. via bash scripts, or other utilities), and (a.3) would feel a bit hacky, since we’d have to deal with config syntax idiosyncracies, and we’d still need a privileged script for accessing the file.

Yeah agreed. I lean towards `settings.yml` because maybe we eventually want to include that file in debug logs.

I definitely don't want to keep state in the Janus config file because that feels too messy.

>(b) How do we write and apply state and configuration changes?
>    (b.1) Extend existing uStreamer launch script to also write the Janus config file and restart the Janus systemd service
>    (b.2) Create new privileged script that’s run as one-off on changes directly from the Python app
>    (b.3) Create our own Janus systemd service, that has its own launcher similar to what we do with uStreamer
>
>In regards to (b), I think it might depend a bit how tight we want to couple uStreamer and Janus on our end. (b.1) would couple tighter but would also better ensure “atomicity”, whereas (b.2) keeps things more separate, and is therefore “cleaner” in terms of code structure. I’m not sure (b.3) is an option, since as far as I’m aware the Janus systemd config is currently not in our control. I’d probably tend towards an independent privileged script for simplicity – like (b.2) –, but I also need to look more into it, and get more familiar with the whole topic. If you have thoughts or preferences here, I’d be curious to hear.

I don't think I have a lot of extra context, but my preference is `b.2`. It just seems simpler to have a standalone script with a simple job as opposed to bundling it with the uStreamer launcher or creating our own systemd service.

>I’m wondering whether we’d even need a separate “advanced” view of the video settings, or if we’d get away with an inlined solution. Below a rough mock-up of what would be possible – just as a starting point for further exploration. Would something like this be an option for us worth to consider, or would you prefer to stash the STUN-related things away in a separate place? (As they might be a bit more “special” compared to the other settings.)

Yeah, I'd definitely like to bury them. Probably 80% of users won't have to use them, and I think less than 3% of users will understand what the settings even mean, so I want to avoid throwing confusing settings at people until they ask.

It's similar to how we bury the "mount mode" for virtual media:

[image]

>The suggested dropdown list is natively supported by browsers, it might also have multiple entries. I’m wondering, however, whether it’s a good idea to hard-code external addresses in our UI. They ultimately aren’t in our control and might change over time. In this case, our UI could get stale, or worse, appears to be broken. If we’d be worried about that, we could consider to e.g. either link to a support page on tinypilotkvm.com with a list of public STUN servers, or we maintain curated addresses via a “static JSON file” hosted at tinypilotkvm.com/stun-server-list.json that we then fetch in the UI.

Let's hardcode it for now for simplicity. If we need something more flexible in the future, we can add a server component, but anytime we bring in client-server interaction, it's like an order of magnitude jump in complexity/maintenance cost.

cghague, 2023-07-24:
Please excuse the unprompted feedback! I note that @mtlynch mentioned placing this under the “Video” settings, but pragmatically it’s a network and privacy setting. Would it therefore make more sense to put it below the “Connection” header under “Security” settings instead?

mtlynch, 2023-08-16:
Cool, this is looking good so far!

>We could consider to only show the input field when the user selects “Custom”, i.e. to hide away the stun.l.google.com:19302 address while “Google” is selected. On the one hand, such a UI would be cleaner, on the other hand, it would become more “bouncy” then, and maybe also less transparent.

Yeah, I'd like to abstract away the `host:port` except for when they choose "Custom." The user won't really care.

>I found [this curated list of public STUN servers](https://gist.github.com/mondain/b0ec1cf5f60ae726202e), but from reviewing it I’m not sure there are other viable candidates for us, that we could provide along with the Google server?

These seem trustworthy:

* `stun.ucsb.edu:3478` (we'd call it "University of California, Santa Barbara")
* `stun.whoi.edu:3478` (we'd call it "Woods Hole Oceanographic Institution")
* `stun.barracuda.com:3478` (we'd call it "Barracuda Networks")
* `stun.gmx.de:3478` - You'd know better than I would. Is GMX a reputable brand in Germany?

### Writing the Janus configs

>The [configure-janus privileged script](https://github.com/tiny-pilot/tinypilot/compare/stun-config?expand=1#diff-0b0f3814acc073fdaab7cbfe643377ada5b20f48bf92f088ba87388b5aca4cd7) would be executed by different users / in two different contexts:

>-    By user root at Debian-package install time (e.g., on a fresh device)
>-    By user tinypilot at runtime

I'm confused by this part. `configure-janus` is a privileged script, so wouldn't it always run as root? The `tinypilot` user [can execute it with `sudo` privileges](https://github.com/tiny-pilot/tinypilot/blob/217d9c7fe6e315e4c151f628d6b2a620dffc389b/debian-pkg/etc/sudoers.d/tinypilot#L3).

mtlynch, 2023-08-21:
With the elimination of 1:1 mapping option, I think we can get rid of the "Connection method" dropdown and just have a single dropdown in the Advanced section called "STUN Server" where the options are:

* None (default)
* Google
* UCSB
* ...

mtlynch, 2023-09-28:
Cool, I think this looks really good!

It's slightly jarring that the control changes when the user selects "Custom," since I think it violates expectations a little of how dropdowns work.

The alternative is that picking "Custom" keeps the dropdown as-is, but it makes a new textbox appear, and that might be more confusing.

I don't think it's a huge deal, and it might be the best option we have, but it did throw me a little bit.

>I’m not sure about the wording of the introductory help text, could you revise that please? 

Sure, here's my suggestion for the text:

```html
<p>A STUN server enables H.264 streaming when your web browser is unable to make a direct network connection to your TinyPilot device.</p>

<a href="https://tinypilotkvm.com/faq/stun-server">Do I need a STUN server?</a>
```

It oversimplifies things a bit, but we can go into more detail in the FAQ.

>Do we already have an online help page that we can link to? E.g., we could extend this one, or create a new one.

I've created a ticket for a new one: https://github.com/tiny-pilot/tinypilotkvm.com/issues/1053

>The list of options in the dropdown seems slightly lengthy to me – do we want to keep all these external providers shown, or shall we boil it down a little bit? E.g., presenting just 2–3, instead of now 5? In addition to the external providers, we’d always have “Disabled” and “Custom”.

>The list of options in the dropdown seems slightly lengthy to me – do we want to keep all these external providers shown, or shall we boil it down a little bit? E.g., presenting just 2–3, instead of now 5? In addition to the external providers, we’d always have “Disabled” and “Custom”.

My thinking is that each one appeals to a slightly different type of user:

* Google: The recognizable, trustworthy choice for most people
* UCSB: For people who are skeptical of tech companies but trusting of academia.
* WHOI: For people who are skeptical of tech companies but trusting of academia.
* Barracuda Networks: For people who are okay with tech companies but distrustful of Google specifically
* GMX: For people who are distrustful of US companies and institutions

Maybe we can cut out WHOI since it's less recognizable than UCSB but serves the same purpose?

I guess the hypothetical user who's skeptical of Google but supports Barracuda is probably pretty rare too.

Let's cut out WHOI and Barracuda.

cghague, 2023-10-16:
@jotaen4tinypilot - Sure thing, I'd be happy to get this tested! For reference, you can test STUN without needing a complex network if you have two Internet connections with a public IPv4 address each. It should be as simple as port forwarding a TinyPilot device on one connection and then accessing it using the other. The H.264 connection should fail at first but work after enabling STUN. This scenario is a typical example of what a customer might face, so it's a good test to try.

@mtlynch - I don't have access to a second connection, so we used a [temporary cloud server](https://github.com/tiny-pilot/tinypilotkvm.com/issues/939#issuecomment-1590028846) as a relay last time. I looked at the Exploding Servers tool, but I don't think the provisioned servers have any ports open to the Internet other than SSH. Can you please set the cloud server back up so I can test this?

mtlynch, 2023-10-17:
@cghague - From my experiments, servers on Exploding Servers already have all ports open. I just tried this:

```bash
sudo apt update && sudo apt install -y ncat
```

Then I tried hitting the IP+ port from my browser, and the request came through fine:

```text
$ nc -l 8585
GET / HTTP/1.1
Host: 51.159.140.152:8585
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/118.0
```

Where did port listening fail for you?

cghague, 2023-10-18:
Thanks @mtlynch, I must admit I didn't check if the ports were open, as it's standard for cloud servers to require users to allow ports explicitly, and I expected the cloud servers created by Exploding Servers would be the same. I couldn't see any options to allow ports in the web interface, so I assumed it still needed to be implemented.

@jotaen4tinypilot - I started working on this but realized that, as the bundle isn't for TinyPilot Pro, it doesn't support user authentication or HTTP encryption. I'm not comfortable exposing devices on my home network directly to the Internet without those features, so could you please provide a TinyPilot Pro build of this for me to test?

mtlynch, 2023-10-18:
@cghague - Can you try generating the Pro bundle? Jan provided [feature branch](https://github.com/tiny-pilot/tinypilot/issues/1460#issuecomment-1765118905). If you merge those changes into a branch on Pro, it will generate a Pro bundle for you.

Keep in mind that we can't link a Pro bundle/build in this thread, as it's public.

cghague, 2023-10-18:
Thanks @mtlynch. I've tried that and successfully merged these changes into a test branch on tinypilot-pro. I've given it a very quick smoke test - it appears to work, and the STUN changes are present. I should be able to test STUN in full tomorrow - I'll let you both know how it goes!

cghague, 2023-10-19:
I've tested this, and the results are as follows:

| Format | STUN | Result |
| - | - | - |
| MJPEG | - | MJPEG works |
| H.264 | Disabled | H.264 fails |
| H.264 | Google | H.264 works |
| H.264 | GMX | H.264 fails |
| H.264 | Custom (`stun.l.google.com:3478`) | H.264 works |
| H.264 | Custom (`<my isp stun server>:3478`) | H.264 fails |

I can see in the service logs that Janus is correctly using both the GMX STUN server and the STUN server provided by my ISP when selected. They return the correct IP address when queried during the Janus startup procedure. This result suggests that the server side is working correctly.

However, the client fails to identify the correct candidate using anything other than the Google STUN servers. I'm unsure why this is, and there's nothing helpful in `about:webrtc` or the browser console.

Is there anything else I can test here?

mtlynch, 2023-10-19:
@cghague - The client and server can use separate STUN servers, right? I'm wondering if there's some kind of client bug that accidentally pins to Google even if we specify a different server, but that wouldn't explain why it breaks if the server uses a different STUN server.

@jotaen4tinypilot - Any ideas?

cghague, 2023-10-19:
I carried on looking into this after my last comment. I had many of the same thoughts as @jotaen4tinypilot, and I've ruled out several possible problems and made a small amount of progress.

I started by setting a few breakpoints using the debugger in the browser's developer tools, and the correct value appears to be getting set and propagated correctly. For a belt-and-braces approach, I also tried intentionally mangling the default value in `janus.js`, which didn't change the outcomes, so we can safely assume that's not the issue.

My ISP markets itself at technical users (they're genuinely excellent and are even ["xkcd/806" compliant](https://xkcd.com/806/)), so I'd be surprised if their STUN server didn't work. However, to be safe, I verified all of the STUN servers I used with the `stuntman-client` tools, and they all responded as I expected. Each STUN server identified the correct external IP address and established a viable port number that WebRTC could use.

I finally tried connecting from Safari on my iPhone, and it _did_ work with the STUN server provided by my ISP. My iPhone uses the same Internet connection as my desktop, so I have no explanation for the different outcomes. I tested Safari on my desktop to rule out a browser issue, and it behaved identically to Firefox (which I used for the earlier tests due to the excellent `about:webrtc` utility).

The above findings suggest that STUN makes a difference and that the changes made by @jotaen4tinypilot work and are configuring Janus correctly. Consequently, I speculate that the failures are likely due to the testing environment we're using. I'll try a few more things tomorrow, including the STUN servers @jotaen4tinypilot suggested.

cghague, 2023-10-24:
Thanks @jotaen4tinypilot! I tested the alternate servers, but initially, none worked, so I dug into this further. I tried:

- Using multiple alternative STUN servers.
  - Only the default Google server worked somewhat reliably.
  -  Even the Google server failed occasionally.
  -  Other servers worked, but only very rarely.
- Trying private browsing mode.
  -  This variable had no noticeable effect.
- Testing with different browsers.
  -  Safari and Firefox both behaved the same.
- Enabling full trickle STUN mode.
  -  This setting shares candidates from the client with the server. According to the Janus documentation, this setting can help if both ends of a WebRTC connection are behind NAT.
  -  There was no noticeable impact.
- Locking Janus to the `tun0` interface.
  -  This change prevented the spurious back channel candidates from appearing but didn't appear to fix the issue overall.
- Disabling host candidates in Firefox.
  -  This change prevented unwanted local host addresses from being tried but didn't help the overall issue.
- Disabling host obfuscation in Firefox.
  -  This change didn't impact the number of successful connections, but it did make it easier for me to identify where the connection failures were happening.
- Connecting the client via a cellular hotspot.
  -  I could not establish any H.264 connectivity over a cellular connection, but a quick speed test revealed horrendously poor ping and throughput rates, so this test probably isn't reliable.
- Disabling IP address tracking prevention in macOS.
  -  This change eliminated the nonsensical candidates being suggested but didn't seem to help obtain valid candidates.
- Testing in a virtual machine.
  -  This test was a false start, as I soon discovered H.264 decoding doesn't work in UTM virtual machines on macOS.

During these tests, I observed the `about:webrtc` page to see where the failures occurred. While there were occasional failures to provide candidate ports on the remote side (i.e., the TinyPilot device), the overwhelming majority were happening on the client side.

Based on my findings, we can deduce that the issue is affecting the client side of the connection, but we don't have enough information to say why. It's plausible that the complex testing environment required to force STUN usage on my home network is causing problems. 

I'd suggest that our next step would be to try and arrange a less complicated STUN testing environment. The ideal would be for us to implement the following basic scenario:

```
TinyPilot -> LAN <- Router (port forward) -> Internet <- Router -> LAN <- Client
```

I don't have access to two suitable connections, so this isn't something I can try at home. Do either of you have the resources to attempt this?

cghague, 2023-10-26:
My take on this is that in scenarios where H.264 already works, it will continue to do so regardless of whether STUN is enabled, and in situations where H.264 doesn't currently work, enabling STUN will either fix the problem or have no effect. It therefore follows that there is no downside to providing the STUN functionality. We should, however, continue investigating this to understand why some servers don't work, as that will allow us to provide better customer support.

mtlynch, 2023-10-26:
Yeah, I think we should move forward with the current implementation. It seems strange that there would be something specific about Janus' client implementation that would make Google the only valid STUN server.

Given the cost of testing the STUN setup and the possibility that the effect we're seeing is just testing error, I think the current implementation is worth moving forward with.

@cghague - Can you share more detail about the debugging steps you took in case we want to re-do that in the future? Was it all through `about:webrtc` or were there other interfaces? I'm curious about how to do these parts:

* Enabling full trickle STUN mode. 
* Locking Janus to the tun0 interface. 
* Disabling host candidates in Firefox. 
* Disabling host obfuscation in Firefox.

cghague, 2023-10-26:
I primarily based my investigations on the output of `about:webrtc` and the Janus logs. The Janus logs are accessible via `journalctl -u janus`. I performed the following actions with information from the Janus logs:

- Confirmed that the STUN server was accessible.
- Checked a valid external IP and port combination was found.
- Read status messages for individual connections.

From the `about:webrtc` tool, the most helpful information was:

- The list of remote candidates.
- The list of local candidates.
- Overall WebRTC connection status.
- The browser's WebRTC logs.

In terms of tweaking the behavior of Janus, I made all changes in `janus.jcfg`. I enabled full trickle mode by adding `full_trickle = true` to the `nat` section and locked Janus to the `tun0` interface by adding `ice_enforce_list = "tun0"`, also to the `nat` section. Note that editing the STUN server via TinyPilot will revert these changes.

Regarding Firefox, the browser allows you to override many STUN/TURN settings in `about:config`. The setting titles are self-explanatory, but given how many are available, you might find it helpful to filter for either `stun` or `ice` to narrow the list down. The two specific options I used were:

- `media.peerconnection.ice.obfuscate_host_addresses`
- `media.peerconnection.ice.no_host`

Any active overrides also conveniently show up in `about:webrtc`.

mtlynch, 2023-10-30:
@cghague - Thanks, that's helpful!