tiny-pilot/tinypilot #1467: Migrate uStreamer directory permissions to the uStreamer Debian package

jdeanwallace, 2023-06-27:
Related https://github.com/tiny-pilot/tinypilot/issues/1353

Set the uStreamer directory permissions via the uStreamer Debian package instead of [via Ansible](https://github.com/tiny-pilot/tinypilot/blob/master/ansible-role-ustreamer/tasks/main.yml#L172-L180).

mtlynch, 2023-07-31:
@db39 - Can you take this on before taking on any new tasks? We've got some Ansible-to-Debian tasks that we need before completing some of the dev work for 2.6.1, so I'd like to prioritize this one, as it will help us test the process of support eng working on this end-to-end.

This is a newly documented process. The dev team has been following kind of ad-hoc processes to do this work for the past year, but we wanted to formalize it more so that the support eng team can participate.

If there are things that are confusing, it's likely a gap in the documentation, so definitely ask about things that don't make sense or places where you get stuck.

db39, 2023-07-31:
@mtlynch - Sure!

mtlynch, 2023-08-01:
We can just repeat the same thing we did in the other script and update the bug to say we're doing it in both places.

db39, 2023-08-01:
Ok, cool! Is there a way to ignore the warning that breaks the build, or do I have to remove the `lintian` task?

mtlynch, 2023-08-01:
We should add a `.lintianignore` line like this:

https://github.com/tiny-pilot/tinypilot/blob/0ef51cafd5d7091f272fc12218982dd9097d72dc/.lintianignore#L13-L15

mtlynch, 2023-08-02:
@db39 - For unexpected CI runs, it's helpful if you can link to the specific run + commit. I think you're talking about this one?

https://github.com/tiny-pilot/tinypilot/commit/11af204a2c8de6f242333e2221805b432646be0c
https://app.circleci.com/pipelines/github/tiny-pilot/tinypilot/3786/workflows/b8933ada-5abe-4a70-ae1e-f68949336fb8

@jdeanwallace - Can you update the instructions? It seems like at the very least `default` should be backticked so the reader knows we mean that specific YAML key.

db39, 2023-08-02:
@mtlynch, @jdeanwallace - Thanks for the info! 

I realized my mistake with the `base-revision` param and fixed it in the following commit: https://github.com/tiny-pilot/tinypilot/commit/b4cc5e46a2d8e25bd6e60cb0d4b1384e411dc3f. It still didn't build a bundle on [this workflow](https://app.circleci.com/pipelines/github/tiny-pilot/tinypilot/3787/workflows/a11a3abb-eccb-41f2-865f-813acd23f0e0).

jdeanwallace, 2023-08-02:
@db39 - Oh, it look like it did build:

Your change: https://github.com/tiny-pilot/tinypilot/commit/b4cc5e46a2d8e25bd6e60cb0d4b1384e411dc3f
Your workflow with a bundle: https://app.circleci.com/pipelines/github/tiny-pilot/tinypilot/3793/workflows/ae32e331-2005-4701-b5b1-227806788be0

db39, 2023-08-02:
@jdeanwallace - yeah that workflow is the one where I used the custom param + manual trigger earlier today.

jdeanwallace, 2023-08-02:
@db39 - Oh right, I see the time difference now.

commit f50547d9 (2023-08-03): Migrate uStreamer directory permissions to the uStreamer Debian package (#1546)

Resolves #1467

This change sets the correct permissions of the uStreamer directory
(i.e., /opt/ustreamer) using the uStreamer Debian package, instead of
using Ansible.

To test this bundle, run:

```bash
curl \
  --silent \
  --show-error \
  --location \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/scripts/install-bundle | \
  sudo bash -s -- \
    https://output.circle-artifacts.com/output/job/e81ad09e-33d3-4703-a812-59bb58f415c2/artifacts/0/bundler/dist/tinypilot-community-20230802T1810Z-1.9 [... cut]
closed 2023-08-03 by db39