tiny-pilot/tinypilot #1482: Upgrade failure when adding janus apt repo

LordAro, 2023-07-05:
## Description

Have Tinypilot Pro 2.5.0 on Buster, am trying to upgrade to 2.5.4

## What's the behavior that you expect?

Upgrade to succeed :)

## What's happening instead?

Running the update (both in web interface and then via ssh) results in the following error:

```
TASK [ansible-role-ustreamer : enable Janus apt suite] *******
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: W:GPG error: http://deb.debian.org/debian buster-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131, E:The repository 'http://deb.debian.org/debian buster-backports InRelease' is not signed."}
```

Looks like debian has changed something about their repo?

Am looking into upgrading to Bullseye in the meantime (but requires a new Pro licence...)

## What are the steps to reproduce this behavior?

Update via web interface, or via `sudo /opt/tinypilot-privileged/update`

## Screenshots

## Logs

mtlynch, 2023-07-05:
Thanks for reporting this, @LordAro!

@db39 - Can you see if you can reproduce this?

db39, 2023-07-05:
@mtlynch - I've managed to [reproduce the error](https://logs.tinypilotkvm.com/iIJRIBLK) on a fresh install of `2.5.0` attempting to update to  `2.5.4`.

Re-flashing the SD card with `2.5.4` or `2.6.0` is a workaround. But I'll take a look to see if I can find the underlying issue.

db39, 2023-07-05:
So the problem is pretty clear from the error message:

```text
Failed to update apt cache: W:GPG error: http://deb.debian.org/debian buster-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131, E:The repository 'http://deb.debian.org/debian buster-backports InRelease' is not signed.
```

`apt` won't install the required `janus` packages from the repository because the repository isn't signed. I'm not sure why the repository is no longer signed - I couldn't find anything after searching online.

After trying a few different methods, one solution is to run the following commands to add the keys using the ubuntu keyserver:

```bash
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0E98404D386FA1D9
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 6ED0E7B82643E131
```

Then run the TinyPilot update through the web UI.

I couldn't find a debian keyserver, but the ubuntu one seems to work in this instance.

closed 2023-08-14 by mtlynch