tiny-pilot/tinypilot #1486: Unable to open database file

rcwang, 2023-07-05:
## Description

Briefly summarize the problem you're experiencing.

## What's the behavior that you expect?

See remote desktop screen when go to https://tinypilot/

## What's happening instead?

Seeing this:

```
{
"code": null,
"message": "unable to open database file"
}
```

## What are the steps to reproduce this behavior?

Don't know how to reproduce. It happens randomly, usually within 2-3 days after a reboot.

## Screenshots

Exactly the same "unable to open database file" message as mentioned above.

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

When the issue occurred, I couldn't SSH nor see remote desktop screen via browser. The below log dump is generated after a reboot when all functionalities are (temporarily) back to normal. I am noticing the errors such as: `fatal: detected dubious ownership in repository at '/opt/tinypilot'` and `fatal: detected dubious ownership in repository at '/opt/ustreamer'` in the dump after reboot. Not sure if that's the root cause.

https://logs.tinypilotkvm.com/xZ6DPqfD

mtlynch, 2023-07-05:
Thanks for reporting this!

Is it possible for you to update to the latest version of TinyPilot Pro or TinyPilot Community? It looks like you're running version 2.3.2, which is 18 months old at this point. We've made a lot of improvements since that release to reduce filesystem access bugs like these.

rcwang, 2023-07-05:
I have not been able to update my tinypilot for a long time. Whenever I run `/opt/tinypilot/scripts/upgrade` it would give me this error: `fatal: detected dubious ownership in repository at '/opt/tinypilot'`

```
[]pilot@tinypilot /opt/tinypilot
$ /opt/tinypilot/scripts/upgrade
+ set -u
+ set -e
++ mktemp -d
+ cd /tmp/tmp.3Upma95ijI
+ git clone https://dev.azure.com/tinypilot/tinypilot-pro/_git/tinypilot-pro .
Cloning into '.'...
remote: Azure Repos
remote: Found 7548 objects to send. (97 ms)
Receiving objects: 100% (7548/7548), 13.85 MiB | 12.77 MiB/s, done.
Resolving deltas: 100% (5024/5024), done.
+ ./quick-install
This script is now DEPRECATED.
Please switch to https://gk.tinypilotkvm.com/get-tinypilot-pro.sh
+ TARGET_VERSION=
+ ALLOW_REINSTALL=false
+ [[ 0 -gt 0 ]]
+ readonly ALLOW_REINSTALL
+ readonly LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ readonly RAMDISK_SIZE_MIB=560
+ RAMDISK_SIZE_MIB=560
++ free --mebi
++ grep --fixed-strings Mem:
++ tr --squeeze-repeats ' '
++ cut --delimiter ' ' --fields 7
+ AVAILABLE_MEMORY_MIB=3382
+ readonly AVAILABLE_MEMORY_MIB
+ INSTALLER_DIR=/mnt/tinypilot-installer
+ trap clean_up EXIT
+ ((  3382 >= 560  ))
+ sudo mkdir /mnt/tinypilot-installer
+ sudo mount --types tmpfs --options size=560m --source tmpfs --target /mnt/tinypilot-installer --verbose
mount: tmpfs mounted on /mnt/tinypilot-installer.
+ readonly INSTALLER_DIR
+ readonly TMPDIR=/mnt/tinypilot-installer/tmp
+ TMPDIR=/mnt/tinypilot-installer/tmp
+ export TMPDIR
+ sudo mkdir /mnt/tinypilot-installer/tmp
+ readonly BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
+ BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
+ readonly TARGET_VERSION_FILE=/mnt/tinypilot-installer/target_version.json
+ TARGET_VERSION_FILE=/mnt/tinypilot-installer/target_version.json
+ readonly TINYPILOT_VERSION_FILE=/opt/tinypilot/VERSION
+ TINYPILOT_VERSION_FILE=/opt/tinypilot/VERSION
+ CURRENT_VERSION=
+ [[ -f /opt/tinypilot/VERSION ]]
+ [[ -d /opt/tinypilot/.git ]]
++ cd /opt/tinypilot/
++ git describe --tags
fatal: detected dubious ownership in repository at '/opt/tinypilot'
To add an exception for this directory, call:

        git config --global --add safe.directory /opt/tinypilot
+ CURRENT_VERSION=
+ clean_up
+ sudo umount --lazy /mnt/tinypilot-installer
+ sudo rm -rf /opt/tinypilot-updater /mnt/tinypilot-installer
```

rcwang, 2023-07-05:
I just realized there's this line saying: `This script is now DEPRECATED.
Please switch to https://gk.tinypilotkvm.com/get-tinypilot-pro.sh`

After executing that script, I am seeing these versions below:

```
Software versions
TinyPilot version: 2.5.0
uStreamer version:  
OS version: Linux tinypilot 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

I will keep an eye on it and report back if it continues to show the same issue after a few days. Thanks a lot for helping me out here!

rcwang, 2023-07-06:
I was trying to update from 2.5.0 to 2.5.4 by clicking on "Systems" then "Update" in the Web UI, however, it failed. Could someone help?

Here's the part of the log that showed fatal messages:

```
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "dpkg-query --show --showformat='${Maintainer}' janus | grep TinyPilot", "delta": "0:00:00.079258", "end": "2023-07-06 00:57:39.609211", "msg": "non-zero return code", "rc": 1, "start": "2023-07-06 00:57:39.529953", "stderr": "dpkg-query: no packages found matching janus", "stderr_lines": ["dpkg-query: no packages found matching janus"], "stdout": "", "stdout_lines": []}
...ignoring
TASK [ansible-role-ustreamer : determine if legacy Janus package is installed] ***
ok: [localhost]
TASK [ansible-role-ustreamer : uninstall legacy Janus package] *****************
skipping: [localhost]
TASK [ansible-role-ustreamer : install apt_key dependency] *********************
ok: [localhost]
TASK [ansible-role-ustreamer : add archive apt repository key] *****************
changed: [localhost]
TASK [ansible-role-ustreamer : enable Janus apt suite] *************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: W:GPG error: http://deb.debian.org/debian buster-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131, E:The repository 'http://deb.debian.org/debian buster-backports InRelease' is not signed."}
```

Full log dump here: https://logs.tinypilotkvm.com/ltTcSxTw

rcwang, 2023-07-06:
After upgrading to `2.5.0`, the issue still persists. It was working fine right after a reboot, but a few hours later when I navigated to `https://tinypilot/`, I saw the same error message that I originally posted:

```
{
    "code": null,
    "message": "unable to open database file"
}
```

Any idea why?

mtlynch, 2023-07-06:
It sounds like you're hitting the same issue as https://github.com/tiny-pilot/tinypilot/issues/1482#issuecomment-1622312117

Can you try this so you can get to 2.5.4?

```bash
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0E98404D386FA1D9
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 6ED0E7B82643E131
```

rcwang, 2023-07-08:
Thanks @mtlynch . I was able to successfully update to 2.5.4 based on your previous comment, however, the `unable to open database file` error still persist in this version.

mtlynch, 2023-07-10:
@db39 - Can you assist here?

db39, 2023-07-10:
From a search of our forum, the solution to the other instances ([instance 1](https://forum.tinypilotkvm.com/-542#post-4), [instance 2](https://forum.tinypilotkvm.com/-524#post-5)) of this `unable to open database file` issue was a [re-flash](https://tinypilotkvm.com/faq/factory-reset).

The earlier issues relating to files (`fatal: detected dubious ownership in repository at '/opt/ustreamer'`) suggest that there are some other file issues on the device - hinting that this is likely a case of filesystem corruption.

I think the best course of action here is to download the latest TinyPilot image and re-flash the SD card. And since this sounds like an older SD card (coming from at least 2.3.2), a new SD card may be beneficial here too.

@rcwang - Are you able to [re-flash](https://tinypilotkvm.com/faq/factory-reset) your TinyPilot's microSD card with the [latest version of TinyPilot](https://tinypilotkvm.com/pro/license-check) and let me know if that solves these issues?

rcwang, 2023-07-11:
Hi @db39 ,

I reflashed a new SD card to the latest TinyPilot community version and plugged it into my Voyager 2. However, I am getting the "No Signal" error when I navigate to the Web UI.

The full log dump is: https://logs.tinypilotkvm.com/fBK5pgoJ

Note the error message below extracted from the log:

```
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.969    stream] -- Using V4L2 device: /dev/video0
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.969    stream] -- Using desired FPS: 0
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.969      http] -- Starting HTTP eventloop ...
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.970    stream] -- H264: Initializing encoder ...
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.977    stream] -- H264: Configuring encoder: DMA=0 ...
Jul 07 18:01:17 mypi launch[561]: ================================================================================
Jul 07 18:01:17 mypi launch[561]: -- ERROR [12.994    stream] -- Can't access device: No such file or directory
Jul 07 18:01:17 mypi launch[561]: -- INFO  [12.994    stream] -- Waiting for the device access ...
Jul 07 18:02:35 mypi launch[561]: -- INFO  [76.281      http] -- HTTP: NEW client (now=1): [192.168.1.161]:45638, id=b576d5fd3534349
Jul 07 20:04:57 mypi launch[561]: -- INFO  [7418.222      http] -- HTTP: DEL client (now=0): [192.168.1.161]:45638, id=b576d5fd3534349, Resource temporarily unavailable (reading,eof)
Jul 10 07:55:26 mypi systemd[1]: ustreamer.service: Main process exited, code=killed, status=9/KILL
Jul 10 07:55:26 mypi systemd[1]: ustreamer.service: Failed with result 'signal'.
```

I listed out the devices on my Voyager 2 using `v4l2-ctl` and below is the output:

```
>[redacted]:/opt/ustreamer-launcher/configs.d $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media3

rpivid (platform:rpivid):
        /dev/video19
        /dev/media2

Cannot open device /dev/video0, exiting.
```

Any help would be appreciated, thank you.

rcwang, 2023-07-11:
After some digging, I found and followed the instructions on this page: https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options#example-tc358743-hdmi-to-csi-capture-chip and it seems to be working now. I can see the remote desktop screen and mouse cursor and keyword works.

db39, 2023-07-12:
@rcwang - thanks for the updates! I'm glad that you're back up and running now.

To help reduce the chances of filesystem corruption in the future, I'd recommend enabling [TinyPilot's read-only filesystem](https://tinypilotkvm.com/faq/read-only-filesystem).

Please feel free to reach out again on the [TinyPilot forum](https://forum.tinypilotkvm.com)  if you have any other issues or questions.

closed 2023-07-12 by db39

mtlynch, 2023-07-13:
@rcwang - Glad you got it working!

Our documentation should explain this better, so I've created #1502