tiny-pilot/tinypilot #1489: Improve CPU throttle lines in TinyPilot logs

mtlynch, 2023-07-05:
The CPU throttle line is currently inconsistent with most other lines in the TinyPilot logs.

We should clean it up to be more user-friendly and consistent with our other lines.

### Current

```text
throttled=0x0
```

### Desired

```text
CPU throttled since boot: No
```

jdeanwallace, 2023-08-02:
While reviewing https://github.com/tiny-pilot/tinypilot/pull/1543, I noticed that the current style of the logs is for the text after the colon to be lowercase. For example:

```
TinyPilot state
Read-only filesystem: off
SSH access: enabled
Mouse jiggler: disabled
```

So in this case, should we use a lowercase "yes"/"no"?

mtlynch, 2023-08-02:
Yeah, that's true. We should stick to lowercase.

PR #1543 (2023-08-09): Improve CPU throttle lines in TinyPilot logs
Resolves #1489

Parses the output of `vcgencmd get_throttled` into a simple “yes” or “no” result.

## Decoding the output of `vcgencmd get_throttled`

We could decode the output of `vcgencmd get_throttled` to get more details, but for our purposes we simply want to detect that nothing relating to throttling has occurred since the device was booted, a situation  which is indicated by an output value of `0x0`.
[image]
commit edcf2f82 (2023-08-09): Improve CPU throttle lines in TinyPilot logs (#1543)

Resolves #1489

Parses the output of `vcgencmd get_throttled` into a simple “yes” or
“no” result.

## Decoding the output of `vcgencmd get_throttled`

We could decode the output of `vcgencmd get_throttled` to get more
details, but for our purposes we simply want to detect that nothing
relating to throttling has occurred since the device was booted, a
situation which is indicated by an output value of `0x0`.
<a data-ca-tag
href="https://codeapprove.com/pr/tiny-pilot/tinypilot/1543"><img
src="https://codeapprove.com/external/github-tag-allbg. [... cut]
closed 2023-08-09 by cghague