tiny-pilot/tinypilot #1496: Install Janus config files from TinyPilot Debian package

mtlynch, 2023-07-07:
We're currently placing Janus' config files using Ansible:

https://github.com/tiny-pilot/tinypilot/blob/d903d32f890505e97f94ec1c4549648037dc3555/ansible-role-ustreamer/tasks/install_janus.yml#L21-L32

It should be trivial to pull this logic out of Ansible and implement it instead in TinyPilot's Debian package.

Some notes on the migration:

* The Ansible task is a `template` task even though the source files don't have any templated values. I don't know if this was a mistake originally or if they used to have templates, but now they're just static files.
  * This makes the task easier, as we should be able to move the files like this:
    * `ansible-role-ustreamer/templates/janus.jcfg.j2` -> `debian-pkg/etc/janus/janus.jcfg`
    * `ansible-role-ustreamer/templates/janus.transport.websockets.jcfg.j2` -> `debian-pkg/etc/janus/janus.transport.websockets.jcfg`
* The Ansible task templatizes the path based on `ustreamer_janus_configs_dir`, but we can hardcode the path to `/etc/janus` for simplicity.

## Similar work

* https://github.com/tiny-pilot/tinypilot/pull/1412

jdeanwallace, 2023-07-21:
@mtlynch - Is this a duplicate of https://github.com/tiny-pilot/ustreamer-debian/issues/12?

Do we want to configure Janus in the TinyPilot or uStreamer Debian package?

mtlynch, 2023-07-24:
>Do we want to configure Janus in the TinyPilot or uStreamer Debian package?

I'd like to do it in the TinyPilot Debian package because then it's all in a single repo. It's maybe more logical from the uStreamer Debian package, but then we're spreading more of our code across repos, which I think outweighs other concerns.

mtlynch, 2023-08-02:
@db39 - Can you take this on next? This one is similar in process to https://github.com/tiny-pilot/tinypilot/issues/1467 but the process is simpler because we can do everything in a single change to the tinypilot repo rather than bouncing between this repo and the ustreamer-debian repo.

mtlynch, 2023-08-03:
@db39 - I don't think we should have to explicitly set the permissions in `postinst`. I think `apt-get` just copies over the files with the right permissions, but double check after installing.

PR #1552 (2023-08-04): Install Janus config files from TinyPilot Debian package
Resolves #1496

This change replaces the Ansible task responsible for creating the Janus config files, implementing it in TinyPilot's Debian package instead.

To test this bundle, run:

```bash
curl \
  --silent \
  --show-error \
  --location \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/scripts/install-bundle | \
  sudo bash -s -- \
    https://output.circle-artifacts.com/output/job/5e8d773d-c381-465e-bd67-425ff99f5cd3/artifacts/0/bundler/dist/tinypilot-community-20230804T1445Z-1.9.0-46+dd77b49.tgz
```
[image]
commit c64840e1 (2023-08-04): Install Janus config files from TinyPilot Debian package (#1552)

Resolves #1496

This change replaces the Ansible task responsible for creating the Janus
config files, implementing it in TinyPilot's Debian package instead.

To test this bundle, run:

```bash
curl \
  --silent \
  --show-error \
  --location \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/scripts/install-bundle | \
  sudo bash -s -- \
    https://output.circle-artifacts.com/output/job/5e8d773d-c381-465e-bd67-425ff99f5cd3/artifacts/0/bundler/dist/tinypilot-community-20230804T1445Z-1.9.0-46+dd77b49.tgz
```
<a d [... cut]
closed 2023-08-04 by db39