tiny-pilot/tinypilot #1576: Investigation: Do we need to specify cma in `/boot/cmdline.txt`?

mtlynch, 2023-08-21:
We currently have logic to make sure [cma=128](https://github.com/tiny-pilot/tinypilot/blob/330978abc7288b64d61f279fd687816f3cfde3f6/ansible-role-ustreamer/tasks/update_boot_cmdline.yml) in `/boot/cmdline.txt`, but it may no longer be necessary. The idea was that CMA needed to be at least 128, but the kernel allocated less than that by default. I've seen mentions on the Pi forums that the default is now higher than 128, so we may be able to cut out this logic entirely.

db39, 2023-08-22:
I did a quick test on a fresh install of `2.6.0`. I ran `dmesg | grep -i cma` and got this result:

```text
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 128 MiB at 0x0000000026c00000
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:FA:C9:8C vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=37b7cc36-02 rootfstype=ext4 fsck.repair=yes cma=128M rootwait
[    0.000000] Memory: 1578496K/1835008K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 125440K reserved, 131072K cma-reserved)
[    5.360473] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    5.369975] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

As expected, TinyPilot reserves 128MiB for CMA.

I removed `cma=128` from `/boot/cmdline.txt` and rebooted. After that, I ran `dmesg | grep -i cma` again and got this result:

```text
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory: 1185280K/1835008K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 125440K reserved, 524288K cma-reserved)
[    5.079070] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    5.098688] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

So it seems that there's a 512MiB CMA memory pool by default (if you don't define your own in `cmdline.txt`).

I also checked this on a fresh install of RaspberryPi OS Lite (32-Bit) (Bullseye) and got the same result:

```text
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory: 1365152K/2019328K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 129888K reserved, 524288K cma-reserved)
[    5.139399] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    5.206513] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

Based on those tests, it appears that the default CMA on RaspberryPi OS Bullseye is 512MiB. Although I haven't been able to find any reference to a default value. From my understanding, we should be able to remove those `ansible` tasks entirely.

mtlynch, 2023-08-22:
Cool that's good news! I've created #1580 for the removal.

closed 2023-08-22 by mtlynch