tiny-pilot/tinypilot #1584: Build Debian package using CircleCI ARM executor

mtlynch, 2023-08-23:
I'm embarrassed it took me this long to realize it, but CircleCI's ARM64 (ARMv8) executors can build ARMv7 (i.e., 32-bit ARM) binaries natively.

We've been using Docker+QEMU to emulate 32-bit ARM on AMD64, but we can just use ARM-native CircleCI instances, skip the emulation, and build Debian packages in about 70s instead of the 4m we were seeing when we use QEMU.

This means we should make the following changes:

* We should build Debian packages on every commit in every branch.
  * We were only using conditional builds because ARMv7 builds were so slow.
* We should stop building AMD64 binaries.
  * We were only building them as a way to catch build failures early because it was too slow to build ARMv7, but if we can build ARMv7 quickly, we have no need for AMD64 binaries.
* We should update `CONTRIBUTING.md` with instructions for building ARMv7 on an AMD64 dev machine.

I created a quick 'n dirty proof of concept here, but it needs a bit of cleanup: https://github.com/tiny-pilot/tinypilot/compare/circleci-arm?expand=1

db39, 2023-08-25:
@mtlynch - I've taken a look at your branch and its changes, and I'm not sure what kind of cleanup you're looking for on  the `circleCI` files.

I'm mostly unfamiliar with the circleCI config files beyond having to change the build branch when doing previous work, so I'm probably just missing some obvious stuff.

We could probably refactor [these lines](https://github.com/tiny-pilot/tinypilot/blob/291d3f441c0162378af9b53e223a7395653c211f/.circleci/continue_config.yml#L143-L163) so that we no longer need to move all packages to a different location (assuming we never want to build AMD64)?

mtlynch, 2023-08-25:
>We could probably refactor [these lines](https://github.com/tiny-pilot/tinypilot/blob/291d3f441c0162378af9b53e223a7395653c211f/.circleci/continue_config.yml#L143-L163) so that we no longer need to move all packages to a different location (assuming we never want to build AMD64)?

Right, we don't need to keep that around since we no longer want AMD64. We can revert that section back to [what it used to look like](https://github.com/tiny-pilot/tinypilot/blob/be727b61e572378830116c3be30841285098a09f/.circleci/config.yml#L86-L92).

When you merge into Pro, it's a little more complicated because we're keeping `microsd_build_branch`. Here's how it should look in Pro (just the .circleci stuff, you can ignore the rest):

https://github.com/tiny-pilot/tinypilot-pro/compare/arm-pro?expand=1

The other piece is adding documentation to `CONTRIBUTING.md` showing how to build ARMv7 from an AMD64 dev machine.

PR #1588 (2023-08-29): Build Debian package using CircleCI ARM executor
Resolves #1584 

We've been using Docker+QEMU to emulate 32-bit ARM on AMD64, but this PR switches to ARM-native CircleCI instances, which skips the emulation.

This achieves a speedup on building Debian packages from about 4m down to about 70s.

Because Debian package building is fast again, we're changing our CI workflow to build a Debian package on every commit rather than just on `master` or just when we manually trigger it.

[image]
commit c381d6f7 (2023-08-29): Build Debian package using CircleCI ARM executor (#1588)

Resolves #1584 

We've been using Docker+QEMU to emulate 32-bit ARM on AMD64, but this PR
switches to ARM-native CircleCI instances, which skips the emulation.

This achieves a speedup on building Debian packages from about 4m down
to about 70s.

Because Debian package building is fast again, we're changing our CI
workflow to build a Debian package on every commit rather than just on
`master` or just when we manually trigger it.

<a data-ca-tag
href="https://codeapprove.com/pr/tiny-pilot/tinypilot/1588"><img
src="https://codeapprove.com [... cut]
closed 2023-08-29 by db39