tiny-pilot/tinypilot #1591: Serial / SSH Console support

tnagels, 2023-08-25:
TinyPilot would become indispensable for network admins if it would have the ability to do reach the console of switches and other networking gear via SSH or a (usb-to-) serial connection.

mtlynch, 2023-08-25:
Thanks for reporting this!

Can you share more about this use-case? What's the scenario you'd like to solve with TinyPilot, and what's currently missing?

tnagels, 2023-09-01:
The use case is to use it for managing networking gear. 

A lot of pro switches, firewalls etc. have console (RS232) access to the CLI. This is used in case the management network interface is not available. Consider it a KVM, but text-based and over a serial port.

How i see the interface is (1) a drop-down at the top where one of all available serial ports can be selected. (2) a terminal window in the web page from which commands can be issued.

The SSH alternative would be helpful to keep everything in one interface. Also it would allow executing SSH from a remote location from which the Tinypilot can be reached.

(Sorry for the late reply.)

cghague, 2023-12-13:
We've also received a [request for serial support](https://secure.helpscout.net/conversation/2436012778/1862?folderId=7223090) in TinyPilot via the support inbox.

cghague, 2023-12-28:
We've received [another request for serial support](https://secure.helpscout.net/conversation/2462628921/1939?folderId=7223090) in TinyPilot via the support inbox.

cghague, 2024-02-14:
We've received [another request for serial support](https://secure.helpscout.net/conversation/2510727512/2130) in TinyPilot via the support inbox.