tiny-pilot/tinypilot #1638: Use semantic HTML for toggle checkboxes

mtlynch, 2023-09-29:
We use a bit of an unusual style our toggle-style HTML checkboxes:

```html
<div class="toggle-container">
  Hide Sensitive Data
  <label class="toggle">
    <input type="checkbox" id="include-sensitive-data" />
    <span class="toggle-slider"></span>
  </label>
</div>
```

Ideally, it would look a little closer to something like:

```html
<label>Hide Sensitive Data</label>
<input type="checkbox" class="toggle-slider" id="include-sensitive-data" />
```

In Playwright, we can't

pipliya, 2023-09-30:
Helle @mtlynch , was checking the issue.
This link is not working for me. 
https://github.com/tiny-pilot/tinypilot-pro/blob/d98885ba9b27b574024a394928eaebe737a9cac5/e2e/security.spec.js#L11-L18

mtlynch, 2023-09-30:
@pipliya - It's in our Pro repo, which is not public, but here's what the link points to:

```javascript
  // This is a workaround for our unusual HTML pattern of styling checkboxes as
  // toggles. We should use a locator based on the checkbox label once we fix
  // https://github.com/tiny-pilot/tinypilot/issues/1638.
  const locateToggle = (cssSelector) => {
    return page.locator(".toggle").filter({ has: page.locator(cssSelector) });
  };

  await locateToggle("#require-authentication").click();
```

I'd like to arrange the HTML so that this Playwright code sets the `<input id="include-sensitive-data">` checkbox in the HTML [above](https://github.com/tiny-pilot/tinypilot/issues/1638#issue-1919636967).

```javascript
await page.getByLabel("Hide Sensitive Data").setChecked(true);
```

pipliya, 2023-09-30:
Got it. Thanks!

mtlynch, 2023-10-23:
@jotaen4tinypilot - Thanks for checking into this!

I thought this would be more straightforward, so given the complexity, it's probably not worth diving into right now. Our current Playwright workaround isn't *too* ugly. 

If/when we pick this back up, I think the most promising path is turning this into a web component.

mtlynch, 2023-10-24:
Ah, right. I meant to do that yesterday. Removed!