tiny-pilot/tinypilot #1651: H.264 failover should tell user to try STUN in Video Settings

mtlynch, 2023-10-06:
Now that we have STUN support in the web UI (see https://github.com/tiny-pilot/tinypilot/issues/1460), we can improve the error message when the user fails over from H.264 to MJPEG because the issue is probably that they need a STUN server:

[image]

mtlynch, 2023-10-19:
This kind of ties into [a PR discussion](https://github.com/tiny-pilot/tinypilotkvm.com/pull/1054#pullrequestreview-1677313842) @cghague and I had a couple of days ago.

My sense is that if H.264 is broken, the chances of STUN fixing it within our userbase is like 90%.

Charles - Am I overlooking other issues that would be likely among our users that would cause the WebRTC connection to fail?

cghague, 2023-10-19:
Interesting question, @mtlynch!

We rarely, if ever, get reports of WebRTC/H.264 failing within a single internal network. Based on that, it's reasonable to say that encoding problems, device configuration, browser compatibility, and other similar issues aren't likely to cause problems for our users.

That only really leaves networking issues as a potential source of problems. It's anecdotal, but my experience suggests restrictive NAT configurations and firewalls are the most common root cause of WebRTC/H.264 failing. STUN can help with both restrictive NAT and some types of firewalls, so I think your interpretation is fair.

mtlynch, 2023-10-26:
How about this text?

>TinyPilot failed to stream video output from your target device using H.264 compression.
>
>You may be able to resolve this issue by using a STUN server. Go to System > Video Settings > Advanced Settings to enable this option.
>
>See System > Logs for diagnostic details.