tiny-pilot/tinypilot #1665: Issue with video - No signal

macbrook98, 2023-10-29:
## Description

I am unable to get video after trying multiple of your solutions. I believe this is related to an error on startup. 

## What's the behavior that you expect?

Video output

## What's happening instead?

Error on startup - Failed to start Load EDID for TC358743 HDMI capture chip
No signal for video feed but keyboard inputs work.

## What are the steps to reproduce this behavior?

1. Start Pi
2. See error

## Screenshots

Unable to capture startup. 

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`
I get response - -bash: /opt/tinypilot/dev-scripts/dump-logs: No such file or directory

macbrook98, 2023-10-29:
I have managed to resolve this issue after following the forum post - https://forum.tinypilotkvm.com/-783/no-signal

sudo apt remove tinypilot

You can then proceed with the [custom installation](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options#example-tc358743-hdmi-to-csi-capture-chip) instructions.

closed 2023-10-29 by macbrook98

mtlynch, 2023-10-30:
Thanks for reporting this! We're working on improved documentation to avoid this issue in https://github.com/tiny-pilot/tinypilot/issues/1662