tiny-pilot/tinypilot #1668: Fix compatibility with Raspberry Pi OS Bookworm

mtlynch, 2023-10-31:
From initial testing, we've [discovered](https://github.com/tiny-pilot/tinypilot-pro/issues/1113) that some of TinyPilot's dependencies fail on Raspberry Pi OS Bookworm.

This is a high-level ticket to track the work of individual work items needed for TinyPilot to achieve Bookworm compatibility.

CHN-STUDENT, 2023-11-03:
Yes, i get error when install it on my pi4 with os bookworm, so i suggest you can add warning on `reamd.me ` which os we need choose?

aetonsi, 2023-12-15:
yes please add to readme that **ONLY RPI OS 32bit bullseye** works..

mtlynch, 2023-12-15:
@CHN-STUDENT, @aetonsi - This information is currently in the README:

https://github.com/tiny-pilot/tinypilot#pre-requisites

aetonsi, 2023-12-19:
> @CHN-STUDENT, @aetonsi - This information is currently in the README:
> 
> https://github.com/tiny-pilot/tinypilot#pre-requisites

Hi yes i saw that, i meant that the addition of the word "only" (with emphasis) would be great

CHN-STUDENT, 2023-12-19:
@mtlynch hey, i mean could you use more highlight and bigger fonts to show this, I think we can see this without wrong personal experience

wsxws, 2024-01-20:
Any news ? (Raspi 5 wont run with Bullseye)

mtlynch, 2024-01-22:
@wsxws - No, no changes. If there are any updates, they'll be posted here or in one of the child tickets listed in the [root bug](https://github.com/tiny-pilot/tinypilot/issues/1668#issue-1970469544).

We welcome any help from users who are interested in taking ownership of some of these tasks so that we can accelerate Bookworm compatibility.

hungry-bogart, 2024-03-20:
I had the same issues with Bookworm, so I rolled back to Bullseye listed as Legacy in the RPI Imager. It hasn't worked for some reason as I ended up getting different dependency issues, though similar to the ones listed above. I plan on working with the scripts. If I have any luck, I'll share what I find.

cghague, 2024-03-20:
Thanks for your message, @hungry-bogart. I'm sorry to hear you're having dependency issues. We currently use Bullseye as our target operating system, so TinyPilot should work correctly on a new Raspberry Pi OS Bullseye installation. Can you please confirm that you performed a fresh installation of both Bullseye and then, subsequently, TinyPilot?

hungry-bogart, 2024-03-20:
Thanks, @cphague. I got it working. I think the dependency issues may have had something to do with a broken pipe issue I got while trying to install TinyPilot over SSH. It may have interrupted dpkg and thus packages were left unconfigured. My fix was just attaching the pi to a proper keyboard and screen to have more stability. Then, I ran sudo dpkg --configure -a and the packages resolved normally.

I have the install running just fine now. I am using Bullseye (Raspberry Pi OS). This is really cool btw. I hope to help test in the future. I see you have a lot of requests at the top of this thread. I'll see if I can help.

cghague, 2024-03-20:
Thanks for the quick response. I'm pleased to hear that your TinyPilot installation is working now! Any help [contributing](https://github.com/tiny-pilot/tinypilot/blob/master/CONTRIBUTING.md) to our development and testing processes would be welcome!

As an aside, I've had similar issues with dropped SSH connections in the past and I've found that using `screen` (or an equivalent) for long-running commands can be a good solution.

sfmcnally, 2025-09-09:
> I was able to successfully install tinypilot on a new raspbian bookworm system.
> 
> 1. Append 'dtoverlay=dwc2,dr_mode=peripheral' to /boot/firmware/config.txt and remove similar config lines.
> 2. Add 'modules-load=dwc2,g_hid' to the kernel cmdline at /boot/firmware/cmdline.txt.
> 3. Install janus from raspbian sources, e.g. via 'apt install janus'
> 4. Download tinypilot from https://gk.tinypilotkvm.com/community/download/latest and unzip it.
> 5. Install the provided yq binary, e.g. via “mv yq /usr/bin/yq && chmod 0755 /usr/bin/yq”.
> 6. Install the provided ustreamer deb package, e.g. via “apt-get install -y ./ustreamer*.deb”.
> 7. Install the provided tinypilot deb package, e.g. via “apt-get install -y ./tinypilot*.deb”. This will result in python dependency errors, which can then be resolved via:
>    'sudo -u tinypilot /bin/bash
>    source /opt/tinypilot/venv/bin/activate
>    python -m ensurepip --upgrade
>    pip install --requirement /opt/tinypilot/requirements.txt'
> 8. Install the provided tinypilot deb package again, e.g. via “apt-get install -y ./tinypilot*.deb”.
> 
> You would have to repeat the same process for updates starting from step 4, as updates via the web UI and some other functions will not work with bookworm.

This also worked for me FWIW. I'm on tailscale so I don't need static IP stuff.

HeldvonKosmos, 2025-10-05:
@sebastianmuell  i use raspberrypi 5 with bookworm 32bit. and tinypilot works thanks to your description, but the mouse and keyboard has no effect. how did you fixed this?
For me the usb-gadget service fails on startup, because it can not access '/dev/hidg0': No such file or directory.

ebarrere, 2025-11-15:
I get that this is not anyone's full-time job, but how is Bookworm still not supported after this many years?? The tinypilot web site claims "enterprise support", so that had better include OS updates before bullseye LTS goes EOL next year...

sebastianmuell, 2025-11-22:
> [@sebastianmuell](https://github.com/sebastianmuell) i use raspberrypi 5 with bookworm 32bit. and tinypilot works thanks to your description, but the mouse and keyboard has no effect. how did you [...]

I have a Raspberry Pi 4B, so your problem may be related to the Pi 5 platform, but I'm not sure.