tiny-pilot/tinypilot #1709: Investigation: Find a more robust way for users to change settings only applied during upgrades or installation

cghague, 2023-12-19:
TinyPilot allows users to customize some settings via text-based configuration files. Scripts within our Debian package parse these files during installation. This approach makes it difficult for users to change these settings at any other time, requiring them to use workarounds such as partially uninstalling TinyPilot.

We should find a better way for users to change these settings. Three possible options could be:

- Make our Debian packages compatible with `dpkg-reconfigure` and update our instructions to match.
- Switch to `apt-get reinstall`, although we've [already decided](https://github.com/tiny-pilot/tinypilot/issues/1662) against this approach.
- Separate the configuration parsing scripts from the installer and run them on-demand or whenever TinyPilot starts.

cghague, 2023-12-20:
There are five methods that I am aware of that would resolve our configuration problems:

- Making our Debian packages compatible with `dpkg-reconfigure`.
- Modifying `install` to uninstall TinyPilot when changes are detected.
- Modifying `install` to use `apt reinstall` instead of `apt install`.
- Separating the configuration scripts and executing them on-demand.
- Separating the configuration scripts and executing them at runtime.

For each of the above methods, I have considered the following factors:

- *Effort* - We could implement some methods within Support Engineering with only minor code changes and documentation updates. In contrast, some of the other methods would require a significant amount of developer time.
- *Risk* - Various risks are associated with changing our installation process. The risk is likely to be greater the more complex the change, particularly if we're working in a technical area with which we don't have experience as a team.
- *Coventionality* - There are standard ways of configuring Linux packages that we may want to match. For example, a typical Linux user would expect an application to parse a configuration file like ours at runtime. This factor has minimal technical impact but affects user experience.
- *Compatibility* - Our current Debian packages are incompatible with some of the standard tooling. This factor is low-impact since we don't publish to a repository and generally recommend users only use our installation scripts rather than directly interacting with `apt` or `dpkg`.

I've summarized my thoughts in the following table:

| Method | Effort | Risk | Conventional | Compatible |
| :- | :- | :- | :- | :- |
| `dpkg-reconfigure` | Hard | High | Yes | Yes |
| `install` detects changes | Easy | Medium | No | No |
| `apt-get reinstall` | Easy | Low | No | No |
| Separate scripts (on-demand) | Medium | Low | No | Possibly indirectly |
| Separate scripts (runtime) | Medium | Low | Yes | Possibly indirectly |

@mtlynch - Based on the above, I'd suggest the best option is to separate the configuration logic into a standalone script, which would be called by the `tinypilot` service each time the application starts.

cghague, 2024-01-25:
@mtlynch - I've provided detailed descriptions of each option in the previous post and updated the summary table so that each option is ranked against the others rather than considered in isolation. I think this illustrates the situation better, but let me know if anything isn't clear!

mtlynch, 2024-01-25:
@cghague - Thanks! This is good progress.

I'd like to bring in the rest of the team for discussion. Can we consolidate the details into the root bug? The thing I'd like to consolidate are:

- [Your full summary from yesterday](https://github.com/tiny-pilot/tinypilot/issues/1709#issuecomment-1908671831)
- Definitions of factors (effort, conventionality, etc.)

I'd also like to flesh out which settings we're considering for this. We should have at least one concrete case based on user requests / support tickets (e.g., user wants to switch from HDMI to USB, user is trying to fix a bad/modified install). We'd probably just want to solve for one common case rather than burn effort on every install config change a user might possibly do.

cghague, 2024-02-26:
@mtlynch - I've updated the original message as requested. Please let me know if you want me to expand or clarify anything!

cghague, 2024-03-18:
Thanks @mtlynch. I've added a section to each method explaining how the user would apply any changes they've made to the TinyPilot settings. I also noticed you'd made a few edits, and while reviewing them, I realized that the "Overview" didn't make sense, so I've updated it to reflect what I think you intended it to say.

Regarding your request to clarify the risks associated with config scripts, I think this might be down to my interpretation of what "code change" actually means. I agree that it would only be a few lines of bash code moved around, but the context of pulling code out of the Debian installer to a standalone file makes that seem like a significant change to me, as we'd have to consider aspects like user permissions and so on.

The impact on user experience is open to interpretation. My reason for thinking it will be drastically different is that, while the user will indeed only run a single script in either scenario, the visible effect of that script will be quite noticeably different. The config script would likely only show a simple confirmation that it has updated the settings, whereas the `get-tinypilot[-pro].sh` script would show an entire installation process.

I hope that makes sense, please let me know if you have any questions!

mtlynch, 2024-03-18:
> Regarding your request to clarify the risks associated with config scripts, I think this might be down to my interpretation of what "code change" actually means. I agree that it would only be a few [...]

@cghague - I agree that these are open to interpretation and subjective. Can you rephrase them to be more objective and specific in factual details?

I think we can disagree about running one bash command over another is a drastically different user experience, but if we explain what the difference in UX is, that's something we'll probably agree on. Similarly, if you think the risky part of refactoring into helper scripts is getting file permissions right, we should say specifically that the risk is getting file permissions right when we refactor.

cghague, 2024-03-20:
Thanks for the clarification, @mtlynch; I've updated the text.

mtlynch, 2024-03-25:
Thanks for updating! I'll pull in the dev team to review once we complete https://github.com/tiny-pilot/tinypilot-pro/issues/1266

mtlynch, 2024-03-27:
@jotaen4tinypilot, @jdeanwallace - Can you review the [root bug description](https://github.com/tiny-pilot/tinypilot/issues/1709#issue-2047840902) and weigh in on which solution seems most sensible to you?

mtlynch, 2024-03-28:
>How likely are we to add support for more of such settings customisations in the future?

It's not likely that we'd ever support an official flow where we guide the user through changing capture hardware.

The goal here is just a slightly tidier way of changing settings that depend on install-time state, the most conspicuous one being capture hardware. The current workaround of `apt remove tinypilot` isn't *too* bad, but it has the downside of temporarily putting the user into a bad state (TinyPilot has been removed) just to change one setting.

>How common is it for users to change these settings? I.e., how big is the portion of users that are interested in that? Do we consider this to be something “normal”, or is it rather somewhat edge-casey and exceptional?

I'll let Charles add more context here since I've stopped following support tickets as closely, but my sense is that it's pretty rare, like a handful of customers per year talk to us about needing to do this.

>For users who are interested to change these settings, how often would they toggle them back and forth? I.e., could we legitimately expect someone to constantly change their keyboard/mouse HID path for some reason, or is it more that users would apply a customisation once and then forget about it?

I don't think people would swap back and forth very often, so it would likely be to change it once and then stick with that change.

mtlynch, 2024-03-28:
>If changing settings is rarely requested, and if we’d consider it bearable for users to go through a full reinstall in that case, I’d lean towards the “Modifying install to use apt reinstall instead of apt install” solution. It’s not super convenient, and it might feel a bit strange to make a full TinyPilot reinstall only to provision custom certificates (or the like), but it’s pretty safe from our side, and we can also transparently communicate what it does.

Just to make sure we're on the same page, for that plan, we wouldn't really be telling users to run `apt reinstall`. We'd be modifying [this line in the `install` script](https://github.com/tiny-pilot/tinypilot/blob/22b3e8ec354861b948a09ac0c8a14bac1e0952a1/bundler/bundle/install#L66) so that it always reinstalls the `tinypilot`, `ustreamer`, and `janus` packages regardless of what version is installed. Is that your understanding?

cghague, 2024-03-29:
> Just to make sure we're on the same page, for that plan, we wouldn't really be telling users to run `apt reinstall`. We'd be modifying [this line in the `install` [...]

That's not quite correct; `apt` will only reinstall the package if exactly the same version is already installed. The flow doesn't change in any other situation. The matrix below, which I've copied from my [comment](https://github.com/tiny-pilot/tinypilot/issues/1662) on the related issue, summarizes this:

| Package state		|  Outcome of `install`	| Outcome of `reinstall` |
| :-					| :-					| :-					 |
| Not installed		|  Install from new		| Install from new		 |
| Installed, outdated	|  Update to latest		| Update to latest		 |
| Installed, up-to-date	|  No action			| Reinstall			 |

mtlynch, 2024-03-29:
@cghague - Is there a difference between "Install from new," "Update to latest," or "Reinstall"? I thought they were all effectively the same thing in that they result in the package in the bundle being installed. 

Could the table be restated like this?

| Package state		|  Outcome of `install`	| Outcome of `reinstall` |
| :-					| :-					| :-					 |
| Not installed		|  Install all packages from bundle		| Install all packages from bundle		 |
| Installed, outdated	|  Install all packages from bundle		 | Install all packages from bundle		 |
| Installed, up-to-date	|  No action			| Install all packages from bundle		 |

jdeanwallace, 2024-03-31:
I agree that ideally we'd probably want to go the `dpkg-reconfigure` route (i.e., solution 1) because that seems like the most Debian-way. However, for now I'd probably go for conditionally executing `install` or `reinstall` based on a "certain condition" (i.e., solution 2 or 3). The proposed solutions suggest that the "certain condition" is determined automatically, but perhaps we could get away with a prettier manually flag.

For example, instead of [this](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options/cb57b9f5c696f596b00358a75416efd1a4d269a8#example-tc358743-hdmi-to-csi-capture-chip):
```bash
(sudo apt remove --yes tinypilot || true) && \
  curl \
    --silent \
    --show-error \
    https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  sudo reboot
```

We could add a flag that allows re-installs if the same version has been detected:
```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
  bash -s -- \
    --allow-reinstall && \
  sudo reboot
```

In `get-tinypilot-pro.sh` we already have an `allow-reinstall` flag:
https://github.com/tiny-pilot/gatekeeper/blob/031632d257a3019baa221ce7d93dbdf11250dcb0/static/get-tinypilot-pro.sh#L23-L25

which we only use for importing old-settings.

So my question is, must the reinstall condition be determined automatically, or can it be indicated by the user?

mtlynch, 2024-04-01:
>However, for now I'd probably go for conditionally executing install or reinstall based on a "certain condition" (i.e., solution 2 or 3)

To clarify, (3) proposes for us to unconditionally do `apt-get reinstall` for all users, not just for certain conditions.

>In get-tinypilot-pro.sh we already have an allow-reinstall flag

I think that the `--allow-reinstall` flag has the same problem we have here. 

I haven't tested it, but I think we'd just see the same behavior. It's a gatekeeper-level check to tell Gatekeeper not to get in the way of the reinstall. It doesn't do anything to prevent `apt-get` from skipping the Debian package install process when the installed version and the target version are the same.

So we can add a flag to `get-tinypilot.sh`, but then we'd have to add plumbing to propagate the flag down to [the `install` script in the bundle](https://github.com/tiny-pilot/tinypilot/blob/65e5b05e699686d859a221e56c747304d229f0a0/bundler/bundle/install#L66-L69) where we actually call `apt-get install`.

jdeanwallace, 2024-04-02:
> To clarify, (3) proposes for us to unconditionally do `apt-get reinstall` for all users, not just for certain conditions.

Thanks for clarifying.

In that case, I would go for option (3) with the possibility of adding a conditional check in `get-tinypilot[-pro].sh` to guard against reinstalling the same version each time.

So essentially, our `./install` script will always attempt a `apt-get install --reinstall`, but the `get-tinypilot[-pro].sh` script would automatically block against reinstalling the same version unless indicated by a CLI flag like `--allow-reinstall`.

This should maintain the current install behaviour while allowing certain users to conditionally reconfigure their device.

mtlynch, 2024-04-02:
Okay, I think we can close this here. It seems like people are comfortable with the `apt-get reinstall` solution. I'd also like to see a scratch implementation of (4) before we move forward with any solution.

For now, let's shelve this since we don't see a lot of customers hitting this issue, but if it begins coming up more, let's revisit this and consider whether it's worth investing more in a cleaner solution than [our current workaround](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options#example-tc358743-hdmi-to-csi-capture-chip) of `(sudo apt remove --yes tinypilot || true)`.

closed 2024-04-02 by mtlynch