tiny-pilot/tinypilot #1729: Pi4B - Mouse / Keyboard BrokenPipeError

strukiii, 2024-01-27:
## Description

I've setup my raspberry pi 4 b (Raspberry Pi OS (Legacy, 32-bit ) Lite) with a usb-c to usb-a cable to the target pc's usb 3.0 port, ran a fresh installation (get-tinypilot.sh) and am seeing video but no mouse / keyboard forwarding.  The other issues mentioning these specific errors have been on different unsupported pis, and the guy using usb-a to usb-a, but I'm using the 4b and the correct cable.  Also, the websocket connection is working correctly verified with devtools.  Also maybe noteworthy, I'm solely relying on power from the usb3.0 port, but am not seeing the 

## What's the behavior that you expect?

Mouse / keyboard emulation

## What's happening instead?

No mouse / keyboard movement

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. move mouse around

## Logs
[tinypilot_log.txt](https://github.com/tiny-pilot/tinypilot/files/14070653/tinypilot_log.txt)

 The dev-scripts folder does not exist. 

 
 ```
carter@raspberrypi:/opt/tinypilot $ ls
app  COPYRIGHT  LICENSE  README.md  requirements.txt  scripts  venv  VERSION
carter@raspberrypi:/opt/tinypilot $ cd scripts
carter@raspberrypi:/opt/tinypilot/scripts $ ls
install-bundle  is-ssh-enabled  render-template  streaming-mode  update-service  upgrade
 ```

strukiii, 2024-01-29:
Update - I've tried the setup on my main PC and it works fine.  The target PC is an acer aspire TC-605 from 2014 so might be causing troubles?

mtlynch, 2024-01-29:
@db39 - Can you see if you can help?

db39, 2024-01-30:
Hi @strukiii, I'm sorry you're running into this input issue.

Thanks for sharing those logs! I can't see anything in there that points to the cause of the issue, unfortunately. 

My first thoughts were that the USB-C to USB-A cable could be one that doesn't support data transfer (which is frustratingly common - e.g., with power-only cables intended for charging phones), which we've seen in the past. But since this works on another machine with the same cable, it's likely a different issue.

Have you tried other USB ports on your Acer machine, particularly front / rear IO? There's a chance there's an issue with some ports on your machine.

If that doesn't work, can you try connecting some other USB device to your Acer to check that it recognizes other USB devices correctly? If you have a USB hub you could test, that would be ideal, since that's how TinyPilot appears to target machines.

Please let me know if you have any questions.

strukiii, 2024-01-30:
Thanks for your time helping me out and looking into it @db39.  Turns out I must've been using a no data transfer usb when testing with the acer, then used a data transfer cable with the main pc.  I used a short one to the acer in hopes of keeping the setup compact, then tested the main pc with a longer data cable. 

New issue on the Acer now that my mouse moves though, sorry for double dipping issues.  Seems to be similar to this  https://github.com/tiny-pilot/tinypilot/issues/1581, but when in windowed mode.  The output should be 1080p?  But the black boxes around the browser window misalign the cursor position. 

https://github.com/tiny-pilot/tinypilot/assets/52466430/682c8847-410b-4cab-a050-27ae412dda6f

closed 2024-01-30 by strukiii