tiny-pilot/tinypilot #1742: Not able to connect/control Windows properly

samkys, 2024-02-10:
I have the earlier version of the TinyPilot which has a Raspberry Pi 4B.

Raspberry Pi 4B was updated to Bullseye Headless Lite. 

Powered via power supply-->USB C.

Data Cable: USB A connected to Windows machine (Microsoft 10 Home, build 19045)

By default, the mouse is the only thing that "works". I narrowed it down to what appears to be how Windows handles these devices but I don't know how to fix it.

If I comment out this line (160): https://github.com/tiny-pilot/tinypilot/blob/master/debian-pkg/opt/tinypilot-privileged/init-usb-gadget#L160
The keyboard works, mouse does not. Leave it alone, uncommented, only mouse works.

Applying these:
`echo 0xEF > bDeviceClass
echo 0x02 > bDeviceSubClass
echo 0x01 > bDeviceProtocol`

from [here](https://irq5.io/2016/12/22/raspberry-pi-zero-as-multiple-usb-gadgets/) did not work but maybe it will give someone more familiar with this more insight?

samkys, 2024-02-10:
I'm open to trying more as in the end I would like to be able to use these in a Windows application. If you have some tests you'd like me to run, just let me know.

cghague, 2024-02-13:
Hi @samkys, thank you for your message. I'm sorry you're having mouse and keyboard support issues.

There could be several explanations for this issue, so we'll need to investigate further. TinyPilot presents itself to your target computer as a composite USB device (essentially a USB hub) with a keyboard and mouse attached, so it's strange for only one to work at a time.

Can you please share a link to your logs? You can obtain the link by going to System, then Logs, then Get Shareable URL.

samkys, 2024-02-13:
Hi @cghague, thank you for your reply. 

Here is the log link: 

https://logs.tinypilotkvm.com/LxLltBqo

samkys, 2024-02-13:
Update: Using this file, the device shows up on Windows and I am able to get input data on Windows for Mouse & Keyboard by sending commands from the terminal in the Pi. This would require some changes on the app side but it does work. The report_desc structure may need some tweaks.

Command for sending a keyboard command:

Send Key "h" to the connected computer:
`echo -ne "\x2\0\0\xb\0\0\0\0\0" > /dev/hidg0`

Clear the send:
`echo -ne "\x2\0\0\0\0\0\0\0\0" > /dev/hidg0`

Command to send a mouse control:
Move the mouse +x 5, +y 5
`echo -ne "\x1\0\x5\x5\0" > /dev/hidg0`

[init-usb-gadget.zip](https://github.com/tiny-pilot/tinypilot/files/14271117/init-usb-gadget.zip)

cghague, 2024-02-13:
Thanks for sending me your logs! I've looked through them, and, as you correctly spotted, one of the `/dev/hidg*` devices that TinyPilot requires is missing, although it's unclear why.

There could be a few explanations for the missing devices, including filesystem corruption and isolated software glitches. Unfortunately, TinyPilot cannot function correctly without both devices being present.

The fact that the network device appears in Device Manager is interesting. It could suggest a configuration error, as the composite USB device can include a network adapter, but it's not a piece of functionality we currently use.

I appreciate it's a hassle, but the quickest way to resolve the issue will likely be a full reinstallation. The logs show that you're running the community version of TinyPilot, so you'd need to flash a fresh copy of [Raspberry Pi OS Bullseye (32-bit)](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy) and [install TinyPilot](https://github.com/tiny-pilot/tinypilot?tab=readme-ov-file#simple-installation) from scratch.

I hope this is helpful! If you have any questions, please let me know.

samkys, 2024-02-13:
Hi @cghague, so that was a fresh install from about ~5mins from before I sent the logs. I had reflashed it to know that it was in a known fresh state before uploading logs to you.

cghague, 2024-02-14:
Thanks for confirming!

A complete reinstallation should have ensured both `/dev/hidg*` devices were present, so this could be a compatibility issue or a hardware fault. It's also possible that other software, either on the TinyPilot device or the target computer, could be interfering.

I appreciate these are basic requests, but it would be helpful to rule out a few potential causes. Could you please connect to a different target machine, ideally using a different USB data cable, and check if you see the same issue?

samkys, 2024-02-14:
Thanks for your time on this, I'm hopeful we can get it all to work.

I don't have a different USB cable available at the moment, so all testing is with the same cable.

With TinyPilot USB Data cable connected to an Ubuntu 20.04 machine, Mouse & Keyboard work as expected. I unplugged from here, and plugged directly into Windows again with the same error, only Mouse works.

Plugged into Windows, the output of `ls /dev/hidg*`:
```
pilot@tinypilot:/opt/tinypilot-privileged $ ls /dev/hidg*
/dev/hidg0  /dev/hidg1
```

cghague, 2024-02-14:
Thanks for confirming!

It sounds like the TinyPilot device is working correctly, so we can assume the issue is with the Windows 10 computer. I can confirm that TinyPilot generally works with Windows 10, so this could be something specific to that particular computer or an isolated glitch.

One possibility is that the Windows 10 machine incorrectly identified the TinyPilot device when it was first connected and didn't fully set up support for all of its capabilities. Can you please try uninstalling/removing the TinyPilot device from Device Manager and see if that helps?

samkys, 2024-02-16:
@cghague thank you for the support! I tried with a different Windows machine and things work, so apparently something is just weird with the first Windows machine I tried.

Again, thank you and I appreciate your time, we can close this ticket.