tiny-pilot/tinypilot #1746: Add SSO functionality to tiny pilot (pro) login procedure

alkl0001, 2024-02-26:
Hi guys,

it would be really nice if SSO-like protocols (e.g. CAS, SAML or OAuth2, etc.) could be implemented.

Thanks,

A.K.

cghague, 2024-02-28:
Hi A.K., thanks for your great suggestion about adding single sign-on functionality to TinyPilot!

We have internal tickets open to investigate integrating with LDAP and enhancing our user permission features. These potential enhancements could pave the way toward SSO support, although it's not yet on our short-term roadmap.

We're always working to improve TinyPilot's software and hardware, and we prioritize features based on user demand, so we'll definitely consider your feedback in our planning.

mtlynch, 2024-02-28:
Closing this, as this would be a feature for our Pro version, not the Community version, but we're tracking your request in our Pro repo!