tiny-pilot/tinypilot #1822: Failed to stream H.264

georgeparissis, 2024-07-12:
## Description

After updating the video settings to H.262, expected the video stream to start using H.262. Instead, it falls back to MJPEG

## What's the behavior that you expect?

To start streaming using H.264

## What's happening instead?

It is streaming using MJPEG

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on 'System->Video Settings'
3. Change to H.264
4. Save

## Screenshots

[image]

## Logs

https://[logs.tinypilotkvm.com/0JGwW0fg](https://logs.tinypilotkvm.com/0JGwW0fg)

db39, 2024-07-12:
Hi @georgeparissis, I'm sorry H.264 streaming isn't working correctly on your TinyPilot.

Thanks for sharing those details and logs.

Are you connecting to your TinyPilot remotely over the internet? Perhaps using a service like Tailscale or remote.it?

georgeparissis, 2024-07-13:
Hi @db39 

Yes, I am connecting over the internet and yes I am using remote.it to connect. It is annoying because when I was back home H.264 was working fine, and now that I am remote it doesn't. It makes the whole experience very slow and bad.

db39, 2024-07-15:
Thanks for confirming, @georgeparissis.

If H.264 works locally, but not when you connect to your TinyPilot remotely over the Internet, it sounds like using a STUN server could help here. In TinyPilot Pro version 2.6.2, we introduced STUN server configuration options in TinyPilot's H.264 video settings to help users running into the same issue.

Your logs suggest your TinyPilot is running TinyPilot Pro version 2.6.1. Could you update to the latest version of TinyPilot Pro (on TinyPilot's web interface, go to System > Update)? Once updated, could you try [configuring a STUN server](https://tinypilotkvm.com/faq/stun-server) to see if that allows you to use H.264 over the Internet?

georgeparissis, 2024-07-16:
Thank you for the help, unfortunately it didn't work.

I tried Google and GMX with no success.

db39, 2024-07-17:
Thanks for the update, @georgeparissis. I'm sorry that didn't work.

Are you able to try [Tailscale](https://tinypilotkvm.com/blog/tailscale) instead of remote.it for a remote connection? It's possible that remote.it isn't suited for H.264 streaming.