tiny-pilot/tinypilot #1830: other Keyboard layouts for pasting?

DustyDiamond, 2024-07-29:
Hi, 
nice work with tinypilot so far. 
I have the problem that I can't really change the keyboard layout/language on my target system. I can only choose between German or en-US/french/netherlands ( all with belgian periods -.- )
So naturally when trying to paste text it gets mingled a lot. Just typing with my host and target set to german works fine, but you can imagine typing 20+ char passwords with special characters is a hassle when you could paste them ;)
So my question: Is it planned or in consideration to extend the paste feature with like a dropdown for other keyboard layouts? Or is there another way to get it to work (prefarably with german layout). I didn't get the chance to look at all the code but it seems tinypilot is interpreting "normal keystrokes" different from what the paste menu does. So maybe it would be sufficient to let the paste menu pretend that what is in the textbox got "typed by the user" since the normal typing in window is working perfectly fine with german layout on host and target. 
Greetings from Germany :)

cghague, 2024-07-29:
Hi @DustyDiamond, thanks for your question about pasting text.

You are correct that the "Paste" functionality only works with the `en-US` and `en-GB` languages. This limitation is because TinyPilot needs to convert the plain text provided into a format that it can send over USB, and this can only be done by looking up each value. This requirement means only languages we have a lookup table for will fully work.

I see that you're a technical user. If you're comfortable working with code, you may be able to add support for your desired language by adding it to the [text-to-hid](https://github.com/tiny-pilot/tinypilot/blob/master/app/text_to_hid.py) lookup tables. Our [contributing guidelines](https://github.com/tiny-pilot/tinypilot/blob/master/CONTRIBUTING.md) are a good starting point for developing with TinyPilot.

I hope this is helpful! Please let me know if you have any questions!

DustyDiamond, 2024-08-02:
Ok so i made some changes in a local clone of the repo and tried to commit to a branch tied to this issue. But the Checks always fail. To be more precise `check-all.py` tries to execute `./dev-scripts/check-for-init-files` which doesnt even exist in the repo, but `./dev-scripts/check-for-init--py-files` does exist. 
Then `./dev-scripts/check-python` tries to execute the coverage command but fails with `coverage: command not found` but when I to execute the pre-commit hook manually coverage runs, still with errors but it runs. ^^ (See picture below) 

[image]

I can't really test the changes, beacause the dev-mode tinypilot wont really show me the pastet contents, but console shows me good hex-values in hid.write. And I can't test it on my Pi4B because I can't commit to a dev-branch to clone onto the pi. 
What would be the best way to move on now? I'm not new to coding but collaboration via git(hub/lab) is pretty new to me :D

cghague, 2024-08-07:
Thanks for getting back to me so quickly! I've looked at your fork on GitHub, and the changes seem great! My only observation is that `MODIFIER_RIGHT_ALT` should function as `MODIFIER_ALT_GR` on a system with a suitable keyboard layout, so there's no need to define it manually.

Assuming you followed [these instructions](https://github.com/tiny-pilot/tinypilot/blob/master/CONTRIBUTING.md#install-packages) to set up your development environment, the `coverage: command not found` error most likely means you've not activated the [Python `venv`](https://docs.python.org/3/library/venv.html) before running the command. You can activate the Python `venv` by running `. venv/bin/activate`.

Regarding the `check-all` issue, I've looked, and [this line](https://github.com/tiny-pilot/tinypilot/blob/master/dev-scripts/check-all#L17) does appear to be a mistake. The actual command appears [on this line](https://github.com/tiny-pilot/tinypilot/blob/master/dev-scripts/check-python#L24) in `check-python` instead. I'll ask our developers to investigate.

DustyDiamond, 2024-08-12:
Hi @cghague , 
I requested a [PR](https://github.com/tiny-pilot/tinypilot/pull/1837) which is adding the German locale Support for pasting. 
Best Regards

cghague, 2024-08-14:
That's awesome, thanks @DustyDiamond!

You should only need the modified build files in your local environment. The only files that we'd need are likely to be:

- `app/text_to_hid.py`
- `app/templates/custom-elements/paste-dialog.html`
- `app/hid/keycodes.py`

Can you please push a new commit that leaves just those files? If all goes well, our continuous integration platform will successfully complete a test build, and I can share this with the team for a complete code review.

Thanks again!

DustyDiamond, 2024-08-14:
Hi @cghague , 
I created a new [PR](https://github.com/tiny-pilot/tinypilot/pull/1842) with only the 3 files necessary. 
Best Regards

cghague, 2024-08-26:
Thanks, @DustyDiamond, I've passed this to our development team!