tiny-pilot/tinypilot #1854: Current installs broken on RPI4 bookworm

gazhay, 2024-11-13:
Using the scripts here [https://github.com/tiny-pilot/tinypilot?tab=readme-ov-file#simple-installation](https://github.com/tiny-pilot/tinypilot?tab=readme-ov-file#simple-installation), install fails with dependency problems under a fresh Rpi4 bookworm install image created with rpi imager.

Tried to do a sudo apt update && sudo apt upgrade to make dependencies appear but sadly they don't.

db39, 2024-11-14:
Hi @gazhay, I'm sorry you ran into this install issue.

As you mentioned, TinyPilot currently only supports Bullseye 32-bit. We're tracking compatibility with Bookworm in [this issue](https://github.com/tiny-pilot/tinypilot/issues/1668).

Our [README already mentions](https://github.com/tiny-pilot/tinypilot?tab=readme-ov-file#pre-requisites) Raspberry Pi OS Bullseye (32-bit) as a pre-requisite for installation. Can I ask how we could improve the language surrounding that? For instance, would changing "Pre-requisites" to "Requirements" help?

gazhay, 2024-11-14:
Either move the pre-requisite section under "simple install" and/or perhaps have a one liner test the arch and codename at install time and spit out a message, rather than attempt to install and fail.

After all, the wording is a bit confusing.

The rpi4 is supported, but the only text is something like "use this simple command to install on any compatible rpi"

When it means, any compatible pi running a supported os.

db39, 2024-11-15:
Thanks, @gazhay!

>perhaps have a one liner test the arch and codename at install time and spit out a message, rather than attempt to install and fail.

I was fairly sure that already existed, so I checked and we do have some [checks to prevent installation on incompatible systems](https://github.com/tiny-pilot/tinypilot/blob/61e914237fd5fce7e131e17a24967bde1ddb62e5/bundler/bundle/install#L21-L40). However, it looks like those checks didn't cover some systems (64-bit), which explains why we saw this behavior.

I've [created a new issue](https://github.com/tiny-pilot/tinypilot/issues/1857) and pushed a fix so that we block installs on 64-bit systems too.

I've also created an issue to [update our README](https://github.com/tiny-pilot/tinypilot/issues/1855) to make it more clear what the requirements are for installation. I've changed the "Pre-requisites" section to a "Requirements" section and moved it into the "Simple installation" heading so it's more visible. Additionally, I've updated the requirement to "A Raspberry Pi 4 running Raspberry Pi OS Bullseye (32-bit)".

I hope that covers everything!

Please let me know if you have any questions or feedback.