tiny-pilot/tinypilot #1863: Question(/possible Feature?): Sending "Power off" keyboard button signal

aetonsi, 2024-12-26:
hi,
i was reading up here: https://github.com/tiny-pilot/tinypilot/wiki/Add-an-RF-controlled-power-socket-to-TinyPilot (😦)
and here: https://forum.tinypilotkvm.com/-29/no-usb-input

... and i was wondering if it would be possible to send the "Power off" keyboard button signal many multimedia keyboards have (not talking about laptops).
I am talking about KBs like these:
<table>
<tbody>
<tr>
<td>[image]</td>
<td>[image]</td>
</tr>
</tbody>
</table>

---

So i stumbled upon this: https://wiki.osdev.org/PS/2_Keyboard
and especially this: https://aeb.win.tue.nl/linux/kbd/scancodes-1.html#power

<details>
<summary>... but not being a linux expert...</summary>
i just wasted many hours trying to find the right device, how to send codes, how to read(?) available codes, and even just how to print "Hi" as in the example code from the <a target="_blank" href="https://forum.tinypilotkvm.com/-29/no-usb-input">thread above</a>:
<pre>
echo -ne "\x20\0\xb\0\0\0\0\0" > /dev/hidg0 && \
echo -ne "\0\0\xc\0\0\0\0\0" > /dev/hidg0 && \
echo -ne "\0\0\0\0\0\0\0\0" > /dev/hidg0
</pre>

This example just prints "H" for me, for some reason.... After many hours i figured out how to print "Hi":
<pre>
echo -ne "\x20\x00\x0B\x00\x00\x00\x00\x00" "\x8B\x00\x00\x00\x00\x00\x00\x00" > /dev/hidg0 && \
echo -ne "\x00\x00\x0C\x00\x00\x00\x00\x00" "\x8C\x00\x00\x00\x00\x00\x00\x00" > /dev/hidg0

# or simply

echo -ne "\x20\0\x0B\0\0\0\0\0" "\x8B\0\0\0\0\0\0\0" "\0\0\x0C\0\0\0\0\0" "\x8C\0\0\0\0\0\0\0" > /dev/hidg0
</pre>
</details>

---

**is this something that could be done? I think it might even be useful to add it to the "visual" keyboard?**

[image]

cghague, 2024-12-30:
Hi @aetonsi, thanks for your great question about adding the Power, Sleep, and Wake keys to TinyPilot.

TinyPilot emulates a USB keyboard and can generate key codes for any key listed in the "Keyboard/Keypad Page" section of the [USB HID Usage Tables](https://usb.org/sites/default/files/hut1_21.pdf). Unfortunately, that doesn't include the Power, Sleep, and Wake keys. The scan codes you found are from older, non-USB keyboards.

USB keyboards that do feature these keys are comparatively rare. As a result, they tend to add the functionality by requiring the user to install custom software or by presenting themselves to the computer as multiple USB devices (likely using the "Generic Desktop Page," although that's speculation on my behalf). We don't have any plans to add this ability to TinyPilot at the moment.

I appreciate that this isn't the answer you wanted, but I hope this context helps and answers your questions.

aetonsi, 2024-12-31:
> I appreciate that this isn't the answer you wanted, but I hope this context helps and answers your questions.

thank you very much, i learned something new anyway!

so if i might also ask, then, the mouse/pointing device is just done by the javascript webpage?

cghague, 2025-01-08:
I'm happy my answer was helpful! Regarding the mouse, TinyPilot calculates the position of the local mouse cursor relative to the video stream of the target computer's screen and sends those coordinates to the target computer as an emulated USB mouse using absolute positioning (akin to how a graphics tablet or touchscreen works). The target computer, the local computer, or both may display a cursor. On the local computer, you can change the appearance of the mouse cursor in TinyPilot for convenience, but the setting has no impact on the target computer. Please let me know if you have any questions!