tiny-pilot/tinypilot #1864: FORCE_INSTALL doesn't work as expected

db39, 2025-01-14:
## Description

Setting `FORCE_INSTALL=1` to force an install when running the simple install commands doesn't work.

## What's the behavior that you expect?

Using `FORCE_INSTALL`  with our simple install commands should allow a user to install TinyPilot community regardless of their platform.

## What's happening instead?

The `FORCE_INSTALL` environment variable isn't set properly, likely because of how the script is called (not preserving the env vars).

```text
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ [[ . = \. ]]
+ SCRIPT_DIR=/opt/tinypilot
+ readonly SCRIPT_DIR
+ readonly TINYPILOT_README=/opt/tinypilot/README.md
+ TINYPILOT_README=/opt/tinypilot/README.md
+ [[ -f /opt/tinypilot/README.md ]]
+ readonly HAS_PRO_INSTALLED
+ [[ 0 = 1 ]]
+ readonly LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ LEGACY_INSTALLER_DIR=/opt/tinypilot-updater
+ readonly RAMDISK_SIZE_MIB=560
+ RAMDISK_SIZE_MIB=560
++ free --mebi
++ grep --fixed-strings Mem:
++ tr --squeeze-repeats ' '
++ cut --delimiter ' ' --fields 7
+ AVAILABLE_MEMORY_MIB=1713
+ readonly AVAILABLE_MEMORY_MIB
+ INSTALLER_DIR=/mnt/tinypilot-installer
+ trap clean_up EXIT
+ ((  1713 >= 560  ))
+ sudo mkdir /mnt/tinypilot-installer
+ sudo mount --types tmpfs --options size=560m --source tmpfs --target /mnt/tinypilot-installer --verbose
mount: tmpfs mounted on /mnt/tinypilot-installer.
+ readonly INSTALLER_DIR
+ readonly TMPDIR=/mnt/tinypilot-installer/tmp
+ TMPDIR=/mnt/tinypilot-installer/tmp
+ export TMPDIR
+ sudo mkdir /mnt/tinypilot-installer/tmp
+ readonly BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
+ BUNDLE_FILE=/mnt/tinypilot-installer/bundle.tgz
++ curl https://gk.tinypilotkvm.com/community/download/latest --location --output /mnt/tinypilot-installer/bundle.tgz --write-out '%{http_code}' --silent
+ HTTP_CODE=200
+ readonly HTTP_CODE
+ [[ 200 != \2\0\0 ]]
+ sudo tar --gunzip --extract --file /mnt/tinypilot-installer/bundle.tgz --directory /mnt/tinypilot-installer
+ sudo chown root:root --recursive /mnt/tinypilot-installer
+ [[ 0 -eq 1 ]]
+ pushd /mnt/tinypilot-installer
/mnt/tinypilot-installer ~
+ sudo TMPDIR=/mnt/tinypilot-installer/tmp ./install
+ grep -q boot=overlay /proc/cmdline
+ [[ -n '' ]]
+ grep --quiet '^Model\s*: Raspberry Pi 4 Model B' /proc/cpuinfo
++ lsb_release --id --short
+ [[ Debian != \R\a\s\p\b\i\a\n ]]
+ echo 'TinyPilot currently only supports the 32-bit version of Raspberry Pi OS 11 "Bullseye".'
TinyPilot currently only supports the 32-bit version of Raspberry Pi OS 11 "Bullseye".
+ echo 'To install TinyPilot, you'\''ll need Raspberry Pi OS 11 "Bullseye" (32-bit).'
To install TinyPilot, you'll need Raspberry Pi OS 11 "Bullseye" (32-bit).
+ exit 1
+ clean_up
+ sudo umount --lazy /mnt/tinypilot-installer
+ sudo rm -rf /opt/tinypilot-updater /mnt/tinypilot-installer
```

## What are the steps to reproduce this behavior?

Using a fresh install of Bullseye 32 bit, run `export FORCE_INSTALL=1` and then our simple install commands:

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    bash - && \
  sudo reboot
```

`FORCE_INSTALL` isn't actually set. So the condition for forcing an install is not evaluated as true, meaning incompatible devices still run through the hardware checks, and TinyPilot fails to install.

We should figure out the exact issue and update our guidance for forcing an install (or modify how we force an install).

db39, 2025-01-14:
Part of the issue is that the env variable isn't set when piping the `curl`ed script into bash. And the other issue is that the env variable isn't set when `get-tinypilot.sh` runs the `install` script either:

```bash
sudo \
  TMPDIR="${TMPDIR}" \
  ./install
```

So we'd effectively need to do something like this in `get-tinypilot.sh` *and* to set `FORCE_INSTALL` when running the commands:

```bash
if [[ -n "${FORCE_INSTALL+x}" ]]; then
  sudo \
    TMPDIR="${TMPDIR}" \
    FORCE_INSTALL=1 ./install
fi

sudo \
  TMPDIR="${TMPDIR}" \
  ./install
```

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
    FORCE_INSTALL=1 bash - && \
  sudo reboot
```

There may be a better way of handling all this, however. 

@cghague Do you have any suggestions?

cghague, 2025-01-16:
@db39 - The bug occurs when we use `sudo` to launch the `install` script. The following should work regardless of whether the user has set the `FORCE_INSTALL` variable:

```bash
sudo --preserve-env=TMPDIR,FORCE_INSTALL ./install
```

The other command works correctly, assuming the user sets the variable with `export`, as the child `bash` process inherits the environment variables of the parent process. I've tested this and can confirm that `get-tinypilot.sh` can access the `FORCE_INSTALL` variable.

I'm not keen on piping `curl` to `bash` like that, but that's another topic!

db39, 2025-01-16:
@cghague - Thanks! I've just tested your suggestion, and it works as you described.

I'll create a PR to update `get-tinypilot.sh` with those changes to preserve those two environment variables for `install`.

PR #1865 (2025-01-17): Fix FORCE_INSTALL when using simple install commands
Resolves #1864 

This change updates `get-install.sh` to preserve the `FORCE_INSTALL` environment variable when running `sudo ./install`.

This allows users to set `FORCE_INSTALL` before running the simple install commands.

To test live on a device, install Raspberry Pi OS Bullseye 32-bit and run:

```bash
export FORCE_INSTALL=1
```

And then:

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/refs/heads/1864-force_install-doesnt-work-as-expected/get-tinypilot.sh | \
    bash - && \
  sudo [... cut]
commit 3921b8e7 (2025-01-17): Fix FORCE_INSTALL when using simple install commands (#1865)

Resolves #1864 

This change updates `get-install.sh` to preserve the `FORCE_INSTALL`
environment variable when running `sudo ./install`.

This allows users to set `FORCE_INSTALL` before running the simple
install commands.

To test live on a device, install Raspberry Pi OS Bullseye 32-bit and
run:

```bash
export FORCE_INSTALL=1
```

And then:

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/refs/heads/1864-force_install-doesnt-work-as-expected/get-tinypilot.sh | \
    bash - && [... cut]
closed 2025-01-17 by db39