tiny-pilot/tinypilot #1879: Update hostname page to make `.local` use clearer

db39, 2025-04-28:
When setting a hostname with `.local` (or similar suffix), TinyPilot reports an error about the format. However, this error can be confusing if users expect to need to set `.local`:

[image]

Can we adjust this page to either:

- Update the error to mention not requiring a `.local` suffix
- Illustrate what the new URL will look like using the hostname in the text box

Here's a super rough mockup of the second suggestion:

[image]

db39, 2025-04-30:
Thanks @cghague! Yeah, I wasn't sure on which direction to take, but it sounds like a clearer prompt or error message is the best way to proceed. I'm leaning towards updating the error message rather than the prompt because this seems like an uncommon scenario and it avoids extra clutter.

db39, 2025-05-13:
@str028, can we update the error message [here](https://github.com/tiny-pilot/tinypilot/blob/a9e7096c25a222741c25060776285e5f6ed43b51/app/templates/custom-elements/change-hostname-dialog.html#L59-L61) to something like:

>Invalid hostname: it can only contain the letters a-z, digits and dashes (it cannot start with a dash, though). It must contain 1-63 characters and cannot be "localhost".  You don't need to include ".local"

scott-tp, 2025-05-21:
@str028  You should have access now.